OpenTabletDriver/OpenTabletDriver #3470: Request for making configurations out of the executable file

RainyLave, 2024-09-17:
### Description

Considering inconvenience when adding configurations to this project, maybe we could put the configuration files out of the executable files so that we can add our  configuration files without recompiling

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-09-17:
This.. already exists!
On windows, create a folder called `Configurations` next to the executables

On Linux, you can do the same or place them in `~/.local/share/OpenTabletDriver/Configurations`

On MacOS, this isnt supported yet, but theres already an issue tracking that.

closed 2024-09-17 by jamesbt365

RainyLave, 2024-09-19:
Tks very much~