OpenTabletDriver/OpenTabletDriver #3474: Support Request: otd-daemon manual stop

Mint-me123, 2024-09-20:
### What do you need help with?

Once in a while my PTH-660 stops using the whole area on the monitor. It is as if the mapping of the tablet and the monitor no longer fits each other.

### What have you tried?

 I use an external monitor with a HDMI switch to 2 other PCs (all linux mint). The only way to get back to normal is a reboot. Nothing I do in the opentabletdriver gui helps. While I have not been able to actively provoke the error I noticed in the FAQ that monitor switch changing etc can upset the settings and the opentabletdriver needs to be restarted. I have tried that without luck. I start the otd-daemon manually and I was wondering it it would make sense to manually stop it and then restart it instead of going through a full reboot. Would that perhaps work and if so how do you stop the otd-daemon I cannot find a command to do so.

### Diagnostic Information

"App Version": "OpenTabletDriver v0.6.4.0+9f4e357910a5917ef86166c195c858c1c65b46d2",
  "Build Date": "12/22/2023",
  "Operating System": {
    "Name": "Linux Mint",
    "Version": "21.3 (Virginia)",
    "Attributes": {
      "ID": "linuxmint",
      "ID_LIKE": "ubuntu debian",
      "PRETTY_NAME": "Linux Mint 21.3",
      "VERSION_ID": "21.3",
      "VERSION_CODENAME": "virginia",
      "KERNEL_VERSION": "5.15.0-119-generic"

"DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/0003:056A:0357.0004/hidraw/hidraw3",
      "Manufacturer": "Wacom Co.,Ltd.",
      "ProductName": "Wacom Intuos Pro M",
      "SerialNumber": "8HQ00A1035112",
      "FriendlyName": "Wacom Intuos Pro M",
      "VendorID": 1386,
      "ProductID": 855,
      "InputReportLength": 192,
      "OutputReportLength": 0,
      "FeatureReportLength": 2561,
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "0",
        "HID_REPORTS": "01:0001:0002, 10:FF0D:0001, 11:FF0D:0001, 13:FF0D:0001, AC:FF0D:0001"

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-09-20:
And if you aren't using the service and you are manually starting it in a terminal, just end the process and start it again? You can exit almost any process started in a terminal by just doing ctrl+c in that terminal.

Otherwise use one of the many task management tools available? There is one in the very desktop environment you are using.

Mint-me123, 2024-09-20:
Thank you I will try that. I thought I had to stop it with a command.

closed 2024-09-20 by jamesbt365