OpenTabletDriver/OpenTabletDriver #3482: Click and tap feature like wacom driver if possible

enesaltinkaya, 2024-09-28:
### Description

Say you are using blender and using a pen button to pan around an object.  
When the button is triggered while the pen is not touching the tablet, movements are not accurate.  
It would be very nice if the pen button triggers only when the pen is touching the tablet.  
Like they do in the wacom driver.  

[image]

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-09-28:
> [x] I have searched the existing issues and this new issue is not a duplicate of any.

[image]

Blatant duplicate of #3165.

closed 2024-09-28 by jamesbt365

enesaltinkaya, 2024-09-28:
Obviously that that ticket will sit there for years to come.

Good to ask once a year :+1: