OpenTabletDriver/OpenTabletDriver #3485: New release based on NET 8.0

westernbarnowl, 2024-10-03:
### Description

I noticed that to compile the project NET 8.0 is used but the old release is still stuck on NET 6.0. This is a problem since major Linux distributions ( Ubuntu 24.04 ) and upcoming ones ( Fedora 41 ) don't include NET 6.0 in their repos anymore because its EOL date which is November of this year is coming soon. I saw that the recommendation for Ubuntu was to install the backports repository but Fedora doesn't have such a repo and it also doesn't solve the issue of using soon to be obsolete software. Therefore I'm asking what the plans of a new release are?

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-10-03:
We already target dotnet 8 on master and 0.6.x and we have a release milestone set up [here](https://github.com/OpenTabletDriver/OpenTabletDriver/milestone/28). But we are limited on resources so it'll when it happens.

gonX, 2025-12-05:
Fixed with #3368

closed 2025-12-05 by gonX