OpenTabletDriver/OpenTabletDriver #3487: Support Request: Use the pen button to scroll/move/zoom the content in Linux

theoarena, 2024-10-05:
### What do you need help with?

In the wacom configuration app for Windows, I can bind a pen button to scroll, pan and zoom content:

[image]

For me, as a designer and full-time tablet user (I don't use a mouse), this behavior is vital and part of my daily routine: scrolling content by just holding the pen button, ctrl + hold the button and scroll to zoom, etc. 
Being unable to zoom in/out on an image (Photoshop, Figma, Illustrator etc) just with the pen makes the workflow very difficult and takes away all the ease of use of the tablet

I've tried everything, but I still haven't been able to replicate this behavior on Linux using OTD.

 

### What have you tried?

- Different linux distros: Ubuntu, Kubuntu, Fedora, Mint, Manjaro etc.
- Reinstalled OTD many times
- Tried different plugins with no success (Scroll bindings gave me hope, but it doesn't work as i expected)
- Different combinations of bindings + absolute/artist mode
- Looked at the repository's 'issues' and found the following requests, which I believe have the same problems as me:
  -  https://github.com/OpenTabletDriver/OpenTabletDriver/issues/1381
  - https://github.com/OpenTabletDriver/OpenTabletDriver/issues/3231

### Diagnostic Information

[diagnostic.json](https://github.com/user-attachments/files/17266542/diagnostic.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

studiomantis, 2025-05-12:
Im looking something like this. I Have a Wacom Intuos Pro 2019, and want to setup the buttons, my im strugling here too.

gonX, 2025-08-27:
FWIW, my drawing software (Krita) does handle pen barrel buttons nicely and allows me to zoom and whatnot. 

This reads a little bit like you're wanting scroll wheel actions with pen buttons - which I can't find an issue tracking yet (so this will be the issue tracking scroll wheel buttons), but it also reads like you want drag scrolling which is #1381

gonX, 2025-12-05:
Mouse scroll bindings have been added for 0.6 with #3840 

If you want drag scroll bindings this is tracked in #1381

thomasaull, 2026-01-19:
@Mrcubix The Artifacts have expired, would you mind triggering the pipeline again?

thomasaull, 2026-01-19:
@Mrcubix Thanks, I’ll give it a try :)