OpenTabletDriver/OpenTabletDriver #3493: [macOS ARM] Window closes when deleting the last element inside textfield

leleoveiga, 2024-10-15:
### Steps to Reproduce

1. Open the app
2. Delete every element inside a textfield (one by one, or by CMD + A)

https://github.com/user-attachments/assets/0a25b978-cb98-4059-830a-a8421e4f2dac

### Expected Behavior

The window remaining open

### Observed Behavior

The window closes

### Tablet Device

Huion H430P

### Diagnostic Information

```json
{
  "App Version": "OpenTabletDriver v0.6.4.0+9f4e357910a5917ef86166c195c858c1c65b46d2",
  "Build Date": "12/22/2023",
  "Operating System": {
    "Name": "Unix",
    "Version": "15.0.1",
    "Attributes": null
  },
  "Environment Variables": {
    "USER": "magalu",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": "/var/folders/xq/8wyhqzvj4pq863dlzsxb8bw00000gn/T/"
  },
  "HID Devices": [
    {
      "DevicePath": "IOService:/AppleARMPE/arm-io@10F00000/AppleH15IO/usb-drd0@2280000/AppleT8122USBXHCI@00000000/usb-drd0-port-hs@00100000/Huion Tablet_H430P@00100000/IOUSBHostInterface@1/AppleUserUSBHostHIDDevice",
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet_H430P",
      "SerialNumber": "",
      "FriendlyName": "Huion Tablet_H430P",
      "VendorID": 9580,
      "ProductID": 100,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true,
      "DeviceAttributes": {
        "HID_REPORTS": "0A:000D:0002"
      }
    },
    {
      "DevicePath": "IOService:/AppleARMPE/arm-io@10F00000/AppleH15IO/usb-drd0@2280000/AppleT8122USBXHCI@00000000/usb-drd0-port-hs@00100000/Huion Tablet_H430P@00100000/IOUSBHostInterface@0/AppleUserUSBHostHIDDevice",
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet_H430P",
      "SerialNumber": "",
      "FriendlyName": "Huion Tablet_H430P",
      "VendorID": 9580,
      "ProductID": 100,
      "InputReportLength": 12,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true,
      "DeviceAttributes": {
        "HID_REPORTS": "08:FF00:0001"
      }
    },
    {
[... 2108 lines cut ...]
      "Message": "Driver is enabled.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2024-10-15T16:24:21.781668-03:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

```

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-10-15:
Duplicate of #2013.

gonX, 2024-10-19:
Will reopen given the context of #2013 is before 0.6.x was deemed necessary - ie. this bug affects stable and should be targeted for the next 0.6.x release.

closed 2024-11-24 by jamesbt365