OpenTabletDriver/OpenTabletDriver #3500: Relative Mode - Convert Area

Syphikor, 2024-10-21:
### Description

Currently the Relative Mode has no way to view or select the tablet area. 

I would like to be able to select the Top, Bottom, Left, Right (Convert Area) in the Relative Mode and view it in that menu Relative mode is almost an empty menu with `X Sensitivity` and `Y Sensitivity`. There are two issue with Relative Mode:
1. It uses the whole tablet area when selected instead of allowing users to select `Width`, `Height`, `X`, `Y` positions on the tablet like in the Absolute Mode along with `X Sensitivity` and `Y Sensitivity`.
2. When i want to swap monitor, i have to swap back to Absolute then Save the changes for multi monitor setups and go back into Relative mode and Save again, as you can imagine this process gets annoying if done often.

tl;dr: Allow users to select and preview and change the functional area of the tablet such as Convert Area (from Absolute Mode) in Relative Mode.

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

Kuuuube, 2024-10-21:
Why would you want to artificially reduce the portion of the tablet that is usable? If you want a higher sensitivity, you can set that. You can also convert in absolute mode and do the math to get the relative mode sens (monitor px / tablet area).

For the monitor switching issue, your requested feature does not fix your issue. Maybe try the Monitor Toggle plugin. It lets you bind a button on your tablet to switch monitors.

affenaffgaffel, 2025-01-25:
> Why would you want to artificially reduce the portion of the tablet that is usable? If you want a higher sensitivity, you can set that. You can also convert in absolute mode and do the math to get [...]

I don't know hy but when I calculate it manually my relative sensitivity is way slower than my absolute one