OpenTabletDriver/OpenTabletDriver #3505: Add support for Huion L610

AECBanana, 2024-10-27:
### Manufacturer

Huion

### Product Name

L610

### Device Width

254mm

### Device Height

158mm

### Pressure Levels

16384

### Lines per Inch (LPI)

5080

### Diagnostic Information

[Huion L610.json](https://github.com/user-attachments/files/17534102/Huion.L610.json)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

AECBanana, 2024-10-27:
[Configurations L610.json](https://github.com/user-attachments/files/17534116/L610.json)

[dump HUION_T23b_240429.txt](https://github.com/user-attachments/files/17534120/HUION_T23b_240429.txt)

AECBanana, 2024-10-29:
udev rules
KERNEL=="uhid",MODE:="0666",OPTIONS+="static_node=uhid"
KERNEL=="uinput",MODE:="0666",OPTIONS+="static_node=uinput"
SUBSYSTEMS=="usb",ATTRS{idVendor}=="256c",MODE:="0666"
SUBSYSTEM=="input", KERNEL=="event*",ATTRS{name}=="HUION 256C PEN STYLUS",MODE:="0666"

gonX, 2025-10-06:
Support added in #3879 and is included in release v0.6.6.0 onwards.

closed 2025-10-06 by gonX