OpenTabletDriver/OpenTabletDriver #3509: Dark mode support and customizable icon in tray

fidoboy, 2024-11-01:
### Description

I'm amazed about how good is this driver. It has saved my old Wacom and now it works flawlessly. However I've two suggestions or requests that may be could make this small piece of art even better:

- Support for dark mode in Windows 10/11
- A better tray icon, similar to the Windows Ink or even better, allow to customize the icon with a user provided one.

Thanks a lot in advance.

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-11-01:
>  I have searched the existing issues and this new issue is not a duplicate of any.

[image]
[image]

And for the icon, define "better". I do not understand why you want to change the tray icon of the driver, so what else is there to make it better?

gonX, 2024-11-01:
There's a few duplicates of this (#3247 #3127 #1740 #1245 etc) but I think we can keep this open given the UI framework change (#2772) is being pushed for internally.

I've converted this issue to just focus on dark mode support, please open another issue (or discussion) about the tray icon.

jamesbt365, 2025-01-24:
Dark mode is within the Avalonia rewrite, which has recently been merged into this repository under the `avalonia` branch.

Closing as that branch will be merged into `master` when its done.

fidoboy, 2025-04-19:
> > I have searched the existing issues and this new issue is not a duplicate of any.
> 
> [image] [image]
> 
> And for the icon, define "better". I do not understand why you want to change the tray icon of the driver, so what else is there to make it better?

"Better" means to follow the Windows aesthetics with a minimalistic symbol.

This feature is already not implemented. Can't understand why @jamesbt365 has closed it as completed

jamesbt365, 2025-04-19:
> This feature is already not implemented. Can't understand why @jamesbt365 has closed it as completed

It actually is! It's on the Avalonia branch, the current main development branch.

closed 2025-12-05 by gonX