OpenTabletDriver/OpenTabletDriver #3512: Wacom GD-0912-U support bug: Eraser input not distinguished from nib input

Gaffen, 2024-11-06:
### Steps to Reproduce

1. Attach device
2. Open tablet debug within OpenTabletDriver
3. Observe that both nib and eraser input are detected as nib input

### Expected Behavior

The eraser should be detected as separate input

### Observed Behavior

The eraser produces the same behavior as the nib

### Tablet Device

Wacom GD-0912-U

### Diagnostic Information

[Wacom_GD-091-U_Debug.json](https://github.com/user-attachments/files/17646298/Wacom_GD-091-U_Debug.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

gonX, 2024-11-23:
Erasers are not supported on MacOS yet. #2894 will likely fix this

The tablet debugger is not too helpful when it comes to tool reports. You need to use the tablet data recording feature to check whether tool reports are parsed, since the driver stack currently expects the output mode (e.g. Artist Mode on Linux) to cache the result of tool reports

gonX, 2025-01-24:
Fixed in #3544 / (v0.6.5.0) and will be forward ported to `master`/Avalonia later.

closed 2025-01-24 by gonX