OpenTabletDriver/OpenTabletDriver #3513: Unable to install rpm package on Fedora 41

jk0893, 2024-11-08:
I think I should mention here that, to make sure it's 100% reproducible, I did it on fresh install, so no other packages or settings would conflict. 

I downloaded an rpm package from releases of 0.6.4.0 version. Then tried installing it with `sudo dnf install ./OpenTabletDriver.rpm` as described in official installation guide (and yes, I made sure that rpm package is in right directory). But then I got an error saying: `nothing provides dotnet-runtime-6.0 needed by opentabletdriver-0.6.4.0-1.x86_64 from @commandline`

Not sure why does it say that `dotnet-runtime-6.0` is even needed since I saw in changelog that you apparently moved to .Net 8
I tried installing `dotnet-runtime-8.0` to see if it would help, but it failed to install with same error anyway.

As it turns out, Fedora 41 doesn't have a package `dotnet-runtime-6.0` and probably won't (since .Net 6 support ends in November of 2024). I hope for your help

jamesbt365, 2024-11-08:
Duplicate of #3510, which touches the same topic as #3485. Also, please use our templates unless you are sure what you are posting does not fit into what our templates cover. This should have been a support request.

closed 2024-11-08 by jamesbt365

jk0893, 2024-11-08:
Thanks for such a quick response! My apologies, didn't notice github search opened issues by default. 

I have another issue, however. Compiled a master branch, got gtk working but after few seconds it says `Unable to connect to the OpenTabletDriver Daemon`. Tried launching Daemon and got an error `No tablets were detected.`
Before that I made sure that neither `wacom` and `hid_uclogic` are running with `sudo rmmod wacom hid_uclogic` command.

Here's what terminal displayed:

[RootHub:Debug]	Initialized internal child hubs: HidSharpDeviceRootHub
[Detect:Info]	Searching for tablets...
[Detect:Debug]	Attempting to match config 'Wacom CTL-472' to device 'CTL-472 (/sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:056A:037A.000F/hidraw/hidraw0)'
[Detect:Debug]	Attempting to match config 'Wacom CTL-472' to device 'CTL-472 (/sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.1/0003:056A:037A.0010/hidraw/hidraw7)'
[Detect:Info]	No tablets were detected.

Sorry for being such a dummy-dummy, I'm pretty much a newbie to linux and know very little.

jamesbt365, 2024-11-08:
You should be compiling your own package like this.
```
git clone https://github.com/OpenTabletDriver/OpenTabletDriver.git --branch 0.6.x
cd OpenTabletDriver
./eng/linux/package.sh --package RedHat --output ./dist/redhat
```

jk0893, 2024-11-08:
Thanks a lot. Tried building with commands you provided, got rpm package installed. But it cannot connect to Daemon? When launched, getting an error saying `Connecting to daemon has timed out. Verify that OpenTabletDriver.Daemon is running`. Not sure what to do

jamesbt365, 2024-11-08:
Follow it like normal (apart from actually installing the package, as thats already done)
https://opentabletdriver.net/Wiki/Install/Linux#fedora

jk0893, 2024-11-08:
Could you, please, be more specific at what you mean by "apart from actually installing the package"? I made sure to run `sudo dracut --regenerate-all --force`. After compiling I got an rpm package. If not installing, what I'm supposed to do with it?

jamesbt365, 2024-11-08:
[image]

jk0893, 2024-11-08:
After running command it says that `opentabletdriver.service does not exist`. Am I actually not supposed to run `sudo dnf install` on that package? My apologies, I feel like a first-grade child right now, sort of embarrassing.

jk0893, 2024-11-08:
Oh, then it explains a lot. This is why I mentioned that "apart from actually installing the package", completely misunderstood you xd. Finally got it working and it actually picked up my tablet now. Much appreciated!