OpenTabletDriver/OpenTabletDriver #3517: Support Request: ugee m708 v2 "linux artist mode" buttons only left click, make the cursor blink and jitter slightly.

clyde254, 2024-11-11:
### What do you need help with?

i'm using artist mode, and the "linux artist mode" pen buttons only left click. mouse buttons only work outside of artist modes. my OS is arch linux x86_64, running gnome 47.1

https://github.com/user-attachments/assets/9e235132-1626-4f1b-9fef-0fe298f38e5d

### What have you tried?

i've restarted my computer, changed the pen's settings in the gnome settings app, changed it's settings in opentabletdriver, with no luck. 

### Diagnostic Information

[bazinga!.json](https://github.com/user-attachments/files/17704355/bazinga.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

khoidauminh, 2024-12-12:
I'm reporting the same left click problem with my Gaomon 1060pro in GNOME Wayland. I haven't seen this issue happen outside of GNOME.

gonX, 2025-08-27:
Mouse buttons are not supported in Artist Mode. Please assign the pen tip to nothing, and the pen buttons to their respectable Linux Artist Mode buttons. Use the Advanced Binding Editor (the set of dots) associated with the binding to show the Artist Mode buttons.

closed 2025-08-27 by gonX

clyde254, 2025-08-27:
Kinda funny you only just got to this. I aint even on arch anymore, and i think the issue was that i was on wayland.