OpenTabletDriver/OpenTabletDriver #3527: Add support for Wacom FT-0405-U0B (very similar to already supported FT-0405-U)

pooispoois, 2024-11-18:
### Manufacturer

Wacom

### Product Name

FT-0405-U0B

### Device Width

127.6

### Device Height

92.8

### Pressure Levels

512

### Additional Information

This tablet is very similar to the already supported FT-0405-U0B. By looking at the hardware and tablet debugger, the only differences seem to be:
- Includes a mouse with two buttons, that trigger the first and second button field in tablet debugger (but not the third).
- The pen seems only to have one button, which triggers the second button in tablet debugger. So button one is never triggered.

The configuration request is mainly to be able to use the included mouse, which right now is not configurable in the interface.

I have redacted the diagnostic information, removing some irrelevant information, in order to fit the maximum comment length.

### Diagnostic Information

```json {
  "App Version": "OpenTabletDriver v0.6.4.0",
  "Build Date": "12/22/2023",
  "Operating System": {
    "Name": "Win32NT",
    "Version": "10.0.19045.0",
    "Attributes": {
      "ServicePack": ""
    }
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\redacted\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\redacted\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\redacted"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#intc816&col01#3&10839a5c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd",
      "Manufacturer": "Unknown Manufacturer",
      "ProductName": "Unknown Product Name",
      "SerialNumber": "",
      "FriendlyName": "Unknown Product Name",
      "VendorID": 32903,
      "ProductID": 2590,
      "InputReportLength": 9,
      "OutputReportLength": 2,
      "FeatureReportLength": 0,
      "CanOpen": false,
      "DeviceAttributes": {
        "HID_REPORTS": "01:0001:0006"
      }
    },
    {
      "DevicePath": "\\\\?\\hid#elan901c&col02#5&211c48ef&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Microsoft",
      "ProductName": "HIDI2C Device",
[... 1460 lines cut ...]
      "Notification": false
    },
    {
      "Time": "2024-11-18T10:55:13.6184985+00:00",
      "Group": "Tablet",
      "Message": "Tablet debugging is enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2024-11-18T10:55:56.0340649+00:00",
      "Group": "Tablet",
      "Message": "Tablet debugging is disabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    }
  ]
}```

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2024-11-18:
I edited your issue a bit because this is the same tablet, that is just garbage at the end of the name that isn't really related to the functionality of this tablet.

What I'm guessing happened here is that OTD did either not support mouse devices at the time or whoever supported it did not have one or was unaware that it does have one.

The parser the tablet is currently using in OpenTabletDriver supports mice, provided it is parsed correctly you should be able to press it on your tablet and see an array of mouse buttons appear on the tablet report.

If everything is parsed correctly, let me know.

jamesbt365, 2024-11-18:
As for the 1 pen button, its probably that way for other pens that this tablet supports, without proper ToolId parsing which i'm not the most familiar with I don't know how to make it "properly" use pen button 1.

gonX, 2024-11-23:
Please check if #3542 fixes your concerns.

gonX, 2024-11-23:
Just noticed your pen button concerns too, please open a separate issue for this as it is unlikely this is the only tablet that is affected by it.

pooispoois, 2024-12-07:
I have confirmed that your fix is effective, and both mouse buttons are now configurable in OpenTabletDriver and functional. Thank you!

closed 2024-12-07 by pooispoois