OpenTabletDriver/OpenTabletDriver #3531: [0.6.x] Tablet Debugger does not work and instead gives a NullReferenceException

gonX, 2024-11-18:
### Steps to Reproduce

1. File -> Reset To Defaults
2. Restart GUI
3. Skip wizard
4. Open Tablet Debugger
5. Receive the NRE popup

### Expected Behavior

No NRE

### Observed Behavior

Has NRE:
```
System.NullReferenceException: Object reference not set to an instance of an object.
   at OpenTabletDriver.UX.Windows.Tablet.TabletDebugger.TabletVisualizer.calculateTabletScale(DigitizerSpecifications digitizer, Single scale) in /home/gonx/prog/_opentabletdriver/opentabletdriver/OpenTabletDriver.UX/Windows/Tablet/TabletDebugger.cs:line 398
   at OpenTabletDriver.UX.Windows.Tablet.TabletDebugger.TabletVisualizer.DrawTouchPoints(Graphics graphics, Single scale, TabletReference tablet) in /home/gonx/prog/_opentabletdriver/opentabletdriver/OpenTabletDriver.UX/Windows/Tablet/TabletDebugger.cs:line 384
   at OpenTabletDriver.UX.Windows.Tablet.TabletDebugger.TabletVisualizer.OnNextFrame(PaintEventArgs e) in /home/gonx/prog/_opentabletdriver/opentabletdriver/OpenTabletDriver.UX/Windows/Tablet/TabletDebugger.cs:line 345
   at OpenTabletDriver.UX.Controls.Generic.ScheduledDrawable.OnPaint(PaintEventArgs e) in /home/gonx/prog/_opentabletdriver/opentabletdriver/OpenTabletDriver.UX/Controls/Generic/ScheduledDrawable.cs:line 29
   at Eto.Forms.Drawable.Callback.OnPaint(Drawable widget, PaintEventArgs e)
   at Eto.GtkSharp.Forms.Controls.DrawableHandler.DrawableConnector.HandleDrawn(Object o, DrawnArgs args)
   at Gtk.Widget.DrawnSignalCallback(IntPtr inst, IntPtr arg0, IntPtr gch)
```

### Tablet Device

PTH-851

### Diagnostic Information

n/a probably but tested on e3b084be6c3e7c2b67549c20cc70ded581d12d8f

### Acknowledgements

- [ ] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [ ] I am on the latest version of OpenTabletDriver.

gonX, 2024-11-18:
initial suspect: 3c0c960e1 from #3138

...and my tablet does not have any touch defined, so that's almost definitely the issue

gonX, 2024-11-18:
> Can't reproduce

It requires a touch tablet, which James missed. As requested on Discord I will now hide these messages.

Mrcubix, 2024-11-18:
Can replicate, most touch tablets don't have any specs defined at the moment, which i believe is addressed mostly in void's config PR?

jamesbt365, 2024-11-24:
Fixed by #3540

PR #3540 (2024-11-23): [0.6.x] Fix tablet debugger touch NRE
Fixes #3531

Also adds some additional nullability checks to avoid future NRE's in this part of the code.

/e: improvements in regards to the hashing code to avoid spamming the user with warnings would be appreciated
closed 2024-11-24 by jamesbt365