OpenTabletDriver/OpenTabletDriver #3533: Plugins not working.

hatwearingdumb, 2024-11-20:
### Steps to Reproduce

1. Open the plugin manager
2. Notice how the plugin list is empty
3. Install plugin from file
4. Get error

### Expected Behavior

Plugin manager lists available plugins, and importing works.

### Observed Behavior

No plugins are displayed, and I get an error upon importing

### Tablet Device

Huion Kamvas Pro 20 (2019)

### Diagnostic Information

{
  "AppVersion": "OpenTabletDriver v0.7.0.0",
  "BuildDate": "",
  "OperatingSystem": {
    "Name": "Fedora Linux",
    "Version": "41 (Workstation Edition)",
    "Attributes": {
      "ID": "fedora",
      "VERSION_ID": "41",
      "VERSION_CODENAME": "",
      "PRETTY_NAME": "Fedora Linux 41 (Workstation Edition)",
      "SUPPORT_END": "2025-12-15",
      "VARIANT": "Workstation Edition",
      "VARIANT_ID": "workstation",
      "KERNEL_VERSION": "6.11.7-300.fc41.x86_64"
    }
  },
  "EnvironmentVariables": {
    "USER": "username",
    "DISPLAY": ":0",
    "WAYLAND_DISPLAY": "wayland-0",
    "PWD": "/home/username",
    "PATH": "~/development/flutter/bin:/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/home/username/.dotnet/tools"
  },
  "Devices": [
    {
      "VendorID": 4152,
      "ProductID": 6220,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "Manufacturer": "SteelSeries",
      "ProductName": "SteelSeries Rival 3",
      "FriendlyName": "SteelSeries Rival 3",
      "SerialNumber": "",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-5/1-5:1.3/0003:1038:184C.0006/hidraw/hidraw5",
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "3",
        "HID_REPORTS_NON_RECONSTRUCTABLE": "true"
      }
    },
    {
      "VendorID": 4152,
      "ProductID": 6220,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "Manufacturer": "SteelSeries",
[... 803 lines cut ...]
      "Time": "2024-11-20T22:38:33.8968737+01:00",
      "Group": "Plugin",
      "Message": "Attempted to load the plugin 1.1.1_release_Kuuube-s-CHATTER-EXTERMINATOR when it is already loaded.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2024-11-20T22:38:33.951709+01:00",
      "Group": "Plugin",
      "Message": "Loading plugin '1.1.1_release_Kuuube-s-CHATTER-EXTERMINATOR'",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    }
  ]
}

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

gonX, 2024-11-20:
You are using `master` - the plugin API is not complete, so no plugins have been written for it yet. Use a release like 0.6.4.0.

closed 2024-11-20 by gonX

hatwearingdumb, 2024-11-20:
Okay, that answers it, thanks!
I am using `master` due to difficulties with installing the .Net 6.0 dependency.

jamesbt365, 2024-11-20:
Use the 0.6.x branch

hatwearingdumb, 2024-11-20:
I have used it before, but I must've forgotten about it after reinstalling it, my bad.