OpenTabletDriver/OpenTabletDriver #3534: pen eraser stays enable

soufhehe, 2024-11-20:
### Steps to Reproduce

OTD v0.6.4.0

open xournalpp
use the back of the pen (eraser)
try using the pen to write, it doesn't work and switch back to the eraser (binded to move in the video below)

https://github.com/user-attachments/assets/f43d364d-c763-47c8-b574-5dcfeb4c6729

### Expected Behavior

I expect the pen not to switch to the eraser

### Observed Behavior

I've tried closing xournalpp but it gets stuck with the eraser and I have to restart OTD for it to work again.

I've tried using 0.7 but I get an error about parsing. Should I open another issue for this ?

### Tablet Device

Wacom PTH-660

### Diagnostic Information

[tablet-data.txt](https://github.com/user-attachments/files/17837497/tablet-data.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

soufhehe, 2024-11-20:
(same as tablet-data.txt above, if you prefer)
```json { 13 64 80 00 00 00 00 00 00 }, Delta:1368,175ms, ReportType:OpenTabletDriver.Plugin.Tablet.DeviceReport
{ 10 40 B0 4D 00 7C 2C 00 00 00 00 00 00 00 00 00 3F 00 00 00 00 00 00 00 00 00 00 }, Delta:1595,325ms, Position:[19888,11388], Pressure:0, PenButtons:[False False], Eraser:False, NearProximity:False, HoverDistance:63, Tilt:[0,0], ReportType:IntuosV2Report
{ 10 40 9B 4D 00 93 2C 00 00 00 00 00 00 00 00 00 3F 00 00 00 00 00 00 00 00 00 00 }, Delta:8,715ms, Position:[19867,11411], Pressure:0, PenButtons:[False False], Eraser:False, NearProximity:False, HoverDistance:63, Tilt:[0,0], ReportType:IntuosV2Report
{ 10 40 9B 4D 00 93 2C 00 00 00 00 00 00 00 00 00 3F 00 00 00 00 00 00 00 00 00 00 }, Delta:10,623ms, Position:[19867,11411], Pressure:0, PenButtons:[False False], Eraser:False, NearProximity:False, HoverDistance:63, Tilt:[0,0], ReportType:IntuosV2Report
{ 10 40 9B 4D 00 93 2C 00 00 00 00 00 00 00 00 00 3F 00 00 00 00 00 00 00 00 00 00 }, Delta:4,893ms, Position:[19867,11411], Pressure:0, PenButtons:[False False], Eraser:False, NearProximity:False, HoverDistance:63, Tilt:[0,0], ReportType:IntuosV2Report
{ 10 40 D8 4C 00 40 2D 00 00 00 00 00 00 00 00 00 3F 00 00 00 00 00 00 00 00 00 00 }, Delta:16,799ms, Position:[19672,11584], Pressure:0, PenButtons:[False False], Eraser:False, NearProximity:False, HoverDistance:63, Tilt:[0,0], ReportType:IntuosV2Report
{ 10 70 8F 4C 00 34 2D 00 00 00 F1 09 00 00 00 00 3F 4D 82 80 71 42 08 10 00 4A 08 }, Delta:3,377ms, Position:[19599,11572], Pressure:0, PenButtons:[False False], Eraser:True, NearProximity:True, HoverDistance:63, Tilt:[-15,9], ReportType:IntuosV2Report
{ 10 40 8F 4C 00 34 2D 00 00 00 F1 09 00 00 00 00 3F 00 00 00 00 00 00 00 00 00 00 }, Delta:14,200ms, Position:[19599,11572], Pressure:0, PenButtons:[False False], Eraser:False, NearProximity:False, HoverDistance:63, Tilt:[-15,9], ReportType:IntuosV2Report
{ 10 40 67 4C 00 A3 2D 00 00 00 00 00 00 00 00 00 3F 00 00 00 00 00 00 00 00 00 00 }, Delta:0,857ms, Position:[19559,11683], Pressure:0, PenButtons:[False False], Eraser:False, NearProximity:False, HoverDistance:63, Tilt:[0,0], ReportType:IntuosV2Report
{ 10 40 57 4C 00 9E 2D 00 00 00 00 00 00 00 00 00 3F 00 00 00 00 00 00 00 00 00 00 }, Delta:0,756ms, Position:[19543,11678], Pressure:0, PenButtons:[False False], Eraser:False, NearProximity:False, HoverDistance:63, Tilt:[0,0], ReportType:IntuosV2Report
{ 10 40 4C 4C 00 9E 2D 00 00 00 00 00 00 00 00 00 3F 00 00 00 00 00 00 00 00 00 00 }, Delta:12,953ms, Position:[19532,11678], Pressure:0, PenButtons:[False False], Eraser:False, NearProximity:False, HoverDistance:63, Tilt:[0,0], ReportType:IntuosV2Report
{ 10 40 45 4C 00 A7 2D 00 00 00 00 00 00 00 00 00 3F 00 00 00 00 00 00 00 00 00 00 }, Delta:0,600ms, Position:[19525,11687], Pressure:0, PenButtons:[False False], Eraser:False, NearProximity:False, HoverDistance:63, Tilt:[0,0], ReportType:IntuosV2Report
{ 10 40 3B 4C 00 B5 2D 00 00 00 00 00 00 00 00 00 3F 00 00 00 00 00 00 00 00 00 00 }, Delta:0,694ms, Position:[19515,11701], Pressure:0, PenButtons:[False False], Eraser:False, NearProximity:False, HoverDistance:63, Tilt:[0,0], ReportType:IntuosV2Report
{ 10 40 32 4C 00 BE 2D 00 00 00 00 00 00 00 00 00 3F 00 00 00 00 00 00 00 00 00 00 }, Delta:11,810ms, Position:[19506,11710], Pressure:0, PenButtons:[False False], Eraser:False, NearProximity:False, HoverDistance:63, Tilt:[0,0], ReportType:IntuosV2Report
{ 10 40 2C 4C 00 CD 2D 00 00 00 00 00 00 00 00 00 3F 00 00 00 00 00 00 00 00 00 00 }, Delta:0,604ms, Position:[19500,11725], Pressure:0, PenButtons:[False False], Eraser:False, NearProximity:False, HoverDistance:63, Tilt:[0,0], ReportType:IntuosV2Report
{ 10 40 25 4C 00 D7 2D 00 00 00 00 00 00 00 00 00 3F 00 00 00 00 00 00 00 00 00 00 }, Delta:1,813ms, Position:[19493,11735], Pressure:0, PenButtons:[False False], Eraser:False, NearProximity:False, HoverDistance:63, Tilt:[0,0], ReportType:IntuosV2Report
{ 10 40 20 4C 00 E4 2D 00 00 00 00 00 00 00 00 00 3F 00 00 00 00 00 00 00 00 00 00 }, Delta:4,140ms, Position:[19488,11748], Pressure:0, PenButtons:[False False], Eraser:False, NearProximity:False, HoverDistance:63, Tilt:[0,0], ReportType:IntuosV2Report
{ 10 40 1B 4C 00 EF 2D 00 00 00 00 00 00 00 00 00 3F 00 00 00 00 00 00 00 00 00 00 }, Delta:9,253ms, Position:[19483,11759], Pressure:0, PenButtons:[False False], Eraser:False, NearProximity:False, HoverDistance:63, Tilt:[0,0], ReportType:IntuosV2Report
{ 10 40 16 4C 00 FC 2D 00 00 00 00 00 00 00 00 00 3F 00 00 00 00 00 00 00 00 00 00 }, Delta:3,312ms, Position:[19478,11772], Pressure:0, PenButtons:[False False], Eraser:False, NearProximity:False, HoverDistance:63, Tilt:[0,0], ReportType:IntuosV2Report
{ 10 40 12 4C 00 05 2E 00 00 00 00 00 00 00 00 00 3F 00 00 00 00 00 00 00 00 00 00 }, Delta:0,944ms, Position:[19474,11781], Pressure:0, PenButtons:[False False], Eraser:False, NearProximity:False, HoverDistance:63, Tilt:[0,0], ReportType:IntuosV2Report
{ 10 70 05 4C 00 19 2E 00 00 00 F0 0B 00 00 00 00 3F 4D 82 80 71 42 08 10 00 4A 08 }, Delta:13,664ms, Position:[19461,11801], Pressure:0, PenButtons:[False False], Eraser:True, NearProximity:True, HoverDistance:63, Tilt:[-16,11], ReportType:IntuosV2Report
{ 10 70 FB 4B 00 2E 2E 00 00 00 F0 0B 00 00 00 00 3F 4D 82 80 71 42 08 10 00 4A 08 }, Delta:0,739ms, Position:[19451,11822], Pressure:0, PenButtons:[False False], Eraser:True, NearProximity:True, HoverDistance:63, Tilt:[-16,11], ReportType:IntuosV2 [... cut ...]
{ 10 78 F3 4A 00 B4 32 00 73 02 F2 06 00 00 00 00 0C 4D 82 80 71 42 08 10 00 4A 08 }, Delta:2,583ms, Position:[19187,12980], Pressure:627, PenButtons:[False False], Eraser:True, NearProximity:True, HoverDistance:12, Tilt:[-14,6], ReportType:IntuosV2Report
{ 10 78 F4 4A 00 B5 32 00 7F 02 F2 06 00 00 00 00 0C 4D 82 80 71 42 08 10 00 4A 08 }, Delta:2,214ms, Position:[19188,12981], Pressure:639, PenButtons:[False False], Eraser:True, NearProximity:True, HoverDistance:12, Tilt:[-14,6], ReportType:IntuosV2Report... (16 KB left)
```

ask if you need more

Kuuuube, 2024-12-04:
I can replicate this on Krita. Except instead of resetting when OTD is restarted, gets stuck forever until I restart my PC.

soufhehe, 2025-01-19:
I installed gentoo and now it works. Does this still for you @Kuuuube ? If not issue can be closed

Kuuuube, 2025-01-19:
Still occurs for me. Running Arch. Libinput tool stays on eraser always after using it once.

gonX, 2025-01-19:
Very likely related: #1579

gonX, 2025-01-21:
Fixed for 0.6.x in #3700

PR #3700 (2025-01-21): Artist Mode: Release opposite tool when changing eraser state
This should help with tablets not capable of reporting out of range getting eraser or tip stuck

Fixes #3534

Alternative to #3695

Needs testing
closed 2025-01-21 by gonX