OpenTabletDriver/OpenTabletDriver #3575: UX-Side Plugin Manager is not instantiated using daemon AppInfo, defaulting to local appdata

Mrcubix, 2024-12-02:
### Steps to Reproduce

1. Build 0.6.x (Either in IDE, or publish it),
2. Create a userdata folder next to the generated Daemon binary,
3. Move the UX away (in the case of running in the IDE, it's already done),
4. UX fetch AppInfo from Daemon,
5. UX proceed to use the Plugin Manager instantiated with the old AppInfo.

### Expected Behavior

Instantiate a new Plugin Manager when the daemon is connected.

### Observed Behavior

Plugin Manager with old path is not updated, as a result Plugins will be loaded from the wrong folder in the UX 
(potentially being outdated, since from a separate install)

### Tablet Device

All

### Diagnostic Information

Daemon: [daemon.json](https://github.com/user-attachments/files/17981782/daemon.json)

[image]

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

Mrcubix, 2024-12-03:
Turns out the preset manager, and probably every part of the UX relying on those old paths are affected

gonX, 2024-12-12:
I'm going to put this in the milestone for 0.6.6 since the bug has likely been around for a while, and unless someone PR's a fix I don't think it's going to make the window for 0.6.5.

Mrcubix, 2024-12-18:
Thinking of adding a setter to PresetManager & PluginManager, so these can be updated when the UX connects to the daemon, which would be the easy way.

The hard way would be to refactor elements so that they stop relying on read-only shared properties.
So yeah easy way and then it might be fixed properly in 0.7 .

Not much relies on the Preset Manager.
The plugin manager on the other hand is used by many controls.
The easiest would probably be to re-instantiate the controls or views making use of it.

I think only `TabletSwitcherPanel` needs to be re-instantiated.

PR #3615 (2025-01-22): [0.6.x] Fix AppInfo desync in UX
Attempt at fixing the Plugin Manager & Preset Manager being instantiated with the wrong AppInfo.

Closes #3575
closed 2025-01-24 by gonX