OpenTabletDriver/OpenTabletDriver #3588: Cursor not moving in application under wine wayland driver

46620, 2024-12-09:
### Steps to Reproduce

1. Enable the [wine-staging wayland driver](https://wiki.archlinux.org/title/Wine#Wayland)
2. Try to move the cursor in game

### Expected Behavior

Cursor moves in application

### Observed Behavior

Cursor does not move in application

[Video](https://youtu.be/To_bLo82W-k)

### Tablet Device

Wacom CTL-4100

### Diagnostic Information

[otd-diag.json](https://github.com/user-attachments/files/18054677/otd-diag.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-12-09:
This is going to be an issue with the wine wayland driver or hyprland and its poor tablet handling, as it is working outside the game, you can see it being passed correctly to the wayland compositor, which needs to handle the rest.

46620, 2024-12-10:
Oh yeah, from more testing it's pointing more towards hyprland being the issue. It originally happened to me on plasma wayland but it seems to justwork™ now outside of hyprland

closed 2024-12-10 by jamesbt365