OpenTabletDriver/OpenTabletDriver #3591: [Linux Artist Mode] All pen buttons registered as left click in GNOME Wayland 47.2

khoidauminh, 2024-12-14:
### Steps to Reproduce

1. Log into GNOME Wayland session.
2. Set the pen buttons
3. Try to use pen buttons (for example, middle click and right click)

### Expected Behavior

Middle/Right click is performed

### Observed Behavior

Left click is performed

### Tablet Device

Gaomon 1060Pro

### Diagnostic Information

[diag.json](https://github.com/user-attachments/files/18135740/diag.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-12-14:
We are just sending regular inputs, it is the desktop environments responsibility on how to handle them, so you probably have to configure them in GNOME or the applications you are attempting to use them in.

khoidauminh, 2024-12-14:
I gave it some more testing. This bug seems to only appear in 47 and does look like a GNOME specific issue since the bug doesn't happen outside of it. I'll open a report over there and see what happens.

closed 2024-12-14 by jamesbt365