OpenTabletDriver/OpenTabletDriver #3606: [Linux] There is no obvious way to disable OpenTabletDriver for certain tablets

khalid151, 2024-12-16:
I have multiple tablets, some of which work better with OpenTabletDriver. I prefer using wacom drivers for wacom tablets.
There's no way (that I'm aware of) to use the two at the same time. There should be a way to blacklist certain tablets in OpenTabletDriver.

zxsa89675, 2024-12-16:
Hello, we tried to solve the issue.

This is what we did:

Added a blacklist feature to allow users to exclude specific tablets from being detected by OpenTabletDriver. This allows users to use other drivers (like Wacom) for certain tablets while still using OpenTabletDriver for others. Modified the Settings class to include a BlacklistedTablets property and updated the Driver class to respect this blacklist during device detection.

You can review changes in this commit: https://github.com/zxsa89675/OpenTabletDriver-OpenTabletDriver-3606/commit/e68c7c482d65d6807fd84cc5e30562b6a7c19700.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

jamesbt365, 2024-12-16:
You would be right that theres no way to disable OpenTabletDriver for certain tablets. As a workaround you can add a configuration override and remove the identifiers for the tablets you don't want OpenTabletDriver to detect.

As you are on linux create `~/.local/share/OpentabletDriver/Configurations`, find your configuration from [here](https://github.com/OpenTabletDriver/OpenTabletDriver/tree/0.6.x/OpenTabletDriver.Configurations/Configurations), copy it into that folder and empty the `DigitizerIdentifiers` and `AuxiliaryDeviceIdentifiers` (if present).

khalid151, 2024-12-16:
Great, that works, thanks.

gonX, 2025-01-24:
Related: #3664

gonX, 2025-01-25:
This is now provided on the `avalonia` branch (however, it is unfortunately broken, which is tracked in #3730)