OpenTabletDriver/OpenTabletDriver #3616: NuGet automated package release fails

gonX, 2024-12-18:
Probably missing API keys?

```ansi
2024-12-18T14:00:46.4119478Z ##[group]Run dotnet pack
2024-12-18T14:00:46.4119741Z [36;1mdotnet pack[0m
2024-12-18T14:00:46.4120059Z [36;1mdotnet nuget push ./nupkg/*.nupkg -s "$SOURCE" -k "$NUGET_AUTH_TOKEN"[0m
2024-12-18T14:00:46.4146327Z shell: /usr/bin/bash -e {0}
2024-12-18T14:00:46.4146539Z env:
2024-12-18T14:00:46.4146699Z   VERSION_SUFFIX: 
2024-12-18T14:00:46.4147148Z   NUGET_AUTH_TOKEN: ***
2024-12-18T14:00:46.4147392Z   SOURCE: https://api.nuget.org/v3/index.json
2024-12-18T14:00:46.4147886Z ##[endgroup]
2024-12-18T14:00:47.0133295Z MSBuild version 17.8.5+b5265ef37 for .NET
2024-12-18T14:00:47.3265591Z   Determining projects to restore...
2024-12-18T14:00:48.3564032Z   Restored /home/runner/work/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver.UX/OpenTabletDriver.UX.csproj (in 449 ms).
2024-12-18T14:00:48.3565905Z   Restored /home/runner/work/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver.csproj (in 440 ms).
2024-12-18T14:00:48.3608615Z   Restored /home/runner/work/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver.Native/OpenTabletDriver.Native.csproj (in 2 ms).
2024-12-18T14:00:48.3708253Z   Restored /home/runner/work/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver.UX.Wpf/OpenTabletDriver.UX.Wpf.csproj (in 469 ms).
2024-12-18T14:00:48.4054196Z   Restored /home/runner/work/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver.Daemon/OpenTabletDriver.Daemon.csproj (in 34 ms).
2024-12-18T14:00:48.4060639Z   Restored /home/runner/work/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver.Desktop/OpenTabletDriver.Desktop.csproj (in 38 ms).
2024-12-18T14:00:48.4119781Z   Restored /home/runner/work/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver.Configurations/OpenTabletDriver.Configurations.csproj (in 3 ms).
2024-12-18T14:00:48.4149708Z   Restored /home/runner/work/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver.Console/OpenTabletDriver.Console.csproj (in 35 ms).
2024-12-18T14:00:48.4178304Z   Restored /home/runner/work/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver.UX.MacOS/OpenTabletDriver.UX.MacOS.csproj (in 504 ms).
2024-12-18T14:00:48.4228295Z   Restored /home/runner/work/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver.Plugin/OpenTabletDriver.Plugin.csproj (in 3 ms).
2024-12-18T14:00:48.4678463Z   Restored /home/runner/work/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver.UX.Gtk/OpenTabletDriver.UX.Gtk.csproj (in 46 ms).
2024-12-18T14:00:48.5309534Z   Restored /home/runner/work/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver.Benchmarks/OpenTabletDriver.Benchmarks.csproj (in 117 ms).
2024-12-18T14:00:48.5649050Z   Restored /home/runner/work/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver.Tests/OpenTabletDriver.Tests.csproj (in 139 ms).
2024-12-18T14:00:48.9460331Z   OpenTabletDriver.Plugin -> /home/runner/work/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver.Plugin/bin/Release/net8.0/OpenTabletDriver.Plugin.dll
2024-12-18T14:00:48.9462705Z   OpenTabletDriver.Native -> /home/runner/work/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver.Native/bin/Release/net8.0/OpenTabletDriver.Native.dll
2024-12-18T14:00:50.2587527Z   Successfully created package '/home/runner/work/OpenTabletDriver/OpenTabletDriver/nupkg/OpenTabletDriver.Plugin.0.6.5.nupkg'.
2024-12-18T14:00:50.2591166Z   Successfully created package '/home/runner/work/OpenTabletDriver/OpenTabletDriver/nupkg/OpenTabletDriver.Native.0.6.5.nupkg'.
2024-12-18T14:00:50.7148610Z   OpenTabletDriver.Configurations -> /home/runner/work/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver.Configurations/bin/Release/net8.0/OpenTabletDriver.Configurations.dll
2024-12-18T14:00:50.9419885Z   OpenTabletDriver -> /home/runner/work/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver/bin/Release/net8.0/OpenTabletDriver.dll
2024-12-18T14:00:51.1614307Z   Successfully created package '/home/runner/work/OpenTabletDriver/OpenTabletDriver/nupkg/OpenTabletDriver.Configurations.0.6.5.nupkg'.
2024-12-18T14:00:51.2194450Z /usr/lib/dotnet/sdk/8.0.111/Sdks/NuGet.Build.Tasks.Pack/buildCrossTargeting/NuGet.Build.Tasks.Pack.targets(221,5): warning NU5104: A stable release of a package should not have a prerelease dependency. Either modify the version spec of dependency "Microsoft.Extensions.DependencyInjection [6.0.0-rc.1.21451.13, )" or update the version field in the nuspec. [/home/runner/work/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver/OpenTabletDriver.csproj]
2024-12-18T14:00:51.2198061Z   Successfully created package '/home/runner/work/OpenTabletDriver/OpenTabletDriver/nupkg/OpenTabletDriver.0.6.5.nupkg'.
2024-12-18T14:00:51.6420428Z Pushing OpenTabletDriver.0.6.5.nupkg to 'https://www.nuget.org/api/v2/package'...
2024-12-18T14:00:51.6528326Z   PUT https://www.nuget.org/api/v2/package/
2024-12-18T14:00:52.5148753Z   Forbidden https://www.nuget.org/api/v2/package/ 860ms
2024-12-18T14:00:52.5187005Z error: Response status code does not indicate success: 403 (The specified API key is invalid, has expired, or does not have permission to access the specified package.).
2024-12-18T14:00:52.6077949Z 
2024-12-18T14:00:52.6077956Z 
2024-12-18T14:00:52.6078224Z Usage: dotnet nuget push [arguments] [options]
2024-12-18T14:00:52.6078487Z 
2024-12-18T14:00:52.6078575Z Arguments:
2024-12-18T14:00:52.6079041Z   [root]  Specify the path to the package and your API key to push the package to the server.
2024-12-18T14:00:52.6079461Z 
2024-12-18T14:00:52.6079541Z Options:
2024-12-18T14:00:52.6079820Z   -h|--help                      Show help information
2024-12-18T14:00:52.6080369Z   --force-english-output         Forces the application to run using an invariant, English-based culture.
2024-12-18T14:00:52.6081595Z   -s|--source <source>           Package source (URL, UNC/folder path or package source name) to use. Defaults to DefaultPushSource if specified in NuGet.Config.
2024-12-18T14:00:52.6082408Z   -ss|--symbol-source <source>   Symbol server URL to use.
2024-12-18T14:00:52.6083020Z   -t|--timeout <timeout>         Timeout for pushing to a server in seconds. Defaults to 300 seconds (5 minutes).
2024-12-18T14:00:52.6083619Z   -k|--api-key <apiKey>          The API key for the server.
2024-12-18T14:00:52.6084349Z   -sk|--symbol-api-key <apiKey>  The API key for the symbol server.
2024-12-18T14:00:52.6085372Z   -d|--disable-buffering         Disable buffering when pushing to an HTTP(S) server to decrease memory usage.
2024-12-18T14:00:52.6086136Z   -n|--no-symbols                If a symbols package exists, it will not be pushed to a symbols server.
2024-12-18T14:00:52.6086908Z   --no-service-endpoint          Does not append "api/v2/package" to the source URL.
2024-12-18T14:00:52.6087839Z   --interactive                  Allow the command to block and require manual action for operations like authentication.
2024-12-18T14:00:52.6088737Z   --skip-duplicate               If a package and version already exists, skip it and continue with the next package in the push, if any.
2024-12-18T14:00:52.6089301Z 
2024-12-18T14:00:52.6204615Z ##[error]Process completed with exit code 1.
```

or raw logs (same as above): https://productionresultssa7.blob.core.windows.net/actions-results/d44b26d3-7ece-49fc-9195-7aece3d82fd4/workflow-job-run-1287b841-589e-5c11-50b2-be5413241f28/logs/job/job-logs.txt?rsct=text%2Fplain&se=2024-12-18T15%3A07%3A48Z&sig=L11V6vJZbW1SkcpAof1KF4pSwsA%2BjYZY8363zpyFQFA%3D&ske=2024-12-18T23%3A20%3A31Z&skoid=ca7593d4-ee42-46cd-af88-8b886a2f84eb&sks=b&skt=2024-12-18T11%3A20%3A31Z&sktid=398a6654-997b-47e9-b12b-9515b896b4de&skv=2024-11-04&sp=r&spr=https&sr=b&st=2024-12-18T14%3A57%3A43Z&sv=2024-11-04

InfinityGhost, 2024-12-18:
I have no clue this is configured correctly afaik

InfinityGhost, 2024-12-18:
Manually pushed the packages. API key expired, maybe we'll push the next release before it expires again.

closed 2024-12-18 by InfinityGhost