OpenTabletDriver/OpenTabletDriver #3620: [0.6.x] Plugins are never disposed of when device is disconnected

Mrcubix, 2024-12-18:
### Steps to Reproduce

1. Create a plugin with some level of persistance, implementing `IDisposable`,
2. Enable said plugin,
3. Disconnect the device,
4. `Elements` in the output mode are not disposed of, the InputDeviceTree is just removed, `Dispose()` is never called.
5. Connect another device or apply settings, only remaining InputDeviceTree see their `Elements` disposed of.

```cs
using OpenTabletDriver.Plugin;
using OpenTabletDriver.Plugin.Attributes;
using OpenTabletDriver.Plugin.Output;
using OpenTabletDriver.Plugin.Tablet;

namespace PluginLeak;

[PluginName(PLUGIN_NAME)]
public sealed class PluginLeakShowcase : IPositionedPipelineElement<IDeviceReport>, IDisposable
{
    public const string PLUGIN_NAME = "Plugin Leak Showcase";
    public static int _lastIdentifier;

    public PluginLeakShowcase()
    {
        _ = Task.Run(Run);
    }

    public event Action<IDeviceReport>? Emit;

    public bool IsRunning { get; private set; }
    public int Identifier { get; private set; }

    public PipelinePosition Position => PipelinePosition.None;

    public void Consume(IDeviceReport value) => Emit?.Invoke(value);

    public async Task Run()
    {
        if (IsRunning) return;

        IsRunning = true;
        Identifier = Interlocked.Increment(ref _lastIdentifier);

        while (IsRunning)
        {
            await Task.Delay(1000);
            Log.Write(PLUGIN_NAME, $"Hello from {PLUGIN_NAME}-{Identifier}");
        }
    }

    public void Dispose()
    {
        IsRunning = false;
    }
}
```

### Expected Behavior

Dispose of Elements when `Disconnected` is invoked

https://github.com/OpenTabletDriver/OpenTabletDriver/blob/ccb32a5b7208a6ced9b84dae12b8e7de20aa2ab5/OpenTabletDriver/Driver.cs#L61-L73

### Observed Behavior

Nothing is done, other than removing the InputDeviceTree.

### Tablet Device

any

### Diagnostic Information

[diag.json](https://github.com/user-attachments/files/18188236/diag.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

Mrcubix, 2024-12-18:
Could prob be solved with something like this

```diff
diff --git a/OpenTabletDriver/Driver.cs b/OpenTabletDriver/Driver.cs
index 7eeaf2a5..8520a168 100644
--- a/OpenTabletDriver/Driver.cs
+++ b/OpenTabletDriver/Driver.cs
@@ -60,6 +60,10 @@ namespace OpenTabletDriver
 
                         tree.Disconnected += (sender, e) =>
                         {
+                            foreach (var obj in tree.OutputMode?.Elements ?? [])
+                                if (obj is IDisposable disposable)
+                                    disposable.Dispose();
+
                             // save the immutable array for later use
                             Unsafe.SkipInit(out ImmutableArray<InputDeviceTree> updatedTrees);
                             ImmutableInterlocked.Update(ref _inputDeviceTrees, (trees, tree) =>
```

On another note, i don't see the point of disposing & re-instantiation of plugins for all tablets when a single one is disconnected.

Mrcubix, 2024-12-22:
Doing a bit a research, and it seem that OTD doesn't dispose of them either on uninstall (and also never actually unload the context).
And the GC definitely isn't going to unload it, so that's probably also leaking.

gonX, 2025-01-24:
Fixed in #3622

PR #3622 (2025-01-22): [0.6.x] Dispose of plugins on InputDeviceTree Disconnected
Not disposing of plugins when a device is disconnected will cause some to stay active.
Plugins still have to properly dispose themselves to avoid a memory leak.

Closes #3620
closed 2025-01-24 by gonX