OpenTabletDriver/OpenTabletDriver #3621: Infected with virus - numerous AV programs

illuculent, 2024-12-18:
### Steps to Reproduce

Try to install update with bitdefender
or 
Goto VirusTotal:

ALYac
Gen:Variant.Tedy.662522
Arcabit
Trojan.Tedy.DA1BFA
BitDefender
Gen:Variant.Tedy.662522
CTX
Zip.unknown.tedy
Emsisoft
Gen:Variant.Tedy.662522 (B)
eScan
Gen:Variant.Tedy.662522
GData
Gen:Variant.Tedy.662522
Trellix (HX)
Gen:Variant.Tedy.662522
VIPRE
Gen:Variant.Tedy.662522

### Expected Behavior

No virus detected

### Observed Behavior

Virus detected, installation prevented, temp downloaded filed deleted by AV Program.

### Tablet Device

Wacom

### Diagnostic Information

No install

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

Kuuuube, 2024-12-18:
This is normal for new releases since OpenTabletDriver is not signed. The program requires sending mouse/keyboard input, accessing USB devices, and connecting to the internet (updater and installing plugins). This makes it a prime target for getting falsely flagged by antiviruses.

I'll be sending emails to the AV companies about this false positive. Reputable AVs usually remove the false flag in less than a week. Thank you for letting us know.

In the meantime, you can add an exception in your antivirus for OpenTabletDriver.

Kuuuube, 2024-12-21:
This is resolved now. All AVs have fixed their false positives.

A few may still show on virustotal due to it taking time for them to push updates and virustotal to get the updates.

closed 2024-12-21 by Kuuuube