OpenTabletDriver/OpenTabletDriver #3626: Community report: Cursor can occasionally teleport to a corner on Linux

gonX, 2024-12-21:
I've seen reports in the community that there is a known bug that causes the cursor to teleport to the bottom right for a single frame. It is very obvious in osu! when using cursor-trail heavy skins.

## Known details
- Does not happen on Windows
- Has only been reported in osu!stable so far
- Happens regardless of artist/absolute output mode
- Earliest time seen is September 2022 with this video, which corresponds to at the very latest OpenTabletDriver 0.6.3.0:
https://github.com/user-attachments/assets/31a1fdec-dfea-4ef3-ad2a-c37da6e6617f (slight volume warning)
- Still happens on OpenTabletDriver 0.6.5.0

It is unknown where in the stack this happens. Given it doesn't happen in Windows, we are likely either doing something wrong with the Linux API, or there is an external bug in udev/libinput.

grhw, 2024-12-29:
I don't know if it's osu! or OTD's fault. But, for now, you can enable `Raw Input` in `Mouse` settings on osu! as a temporary fix.

gonX, 2024-12-29:
If that is a consistent workaround, it suggests that WINE could be at fault here. Anyone having this issue should add what WINE version they are using.