OpenTabletDriver/OpenTabletDriver #3627: Support Request: Genius MousePen i608x

KnoyanMitsu, 2024-12-22:
### What do you need help with?

OpenTabletDriver support Genius i608x but weird thing my Genius MousePen i608x not detecting on OTD

Here the information:
```
...
Dec 22 12:00:47 knoyan.local otd-daemon[4024]: [Detect:Debug]        Searching for tablet 'Gaomon S830'
Dec 22 12:00:47 knoyan.local otd-daemon[4024]: [Detect:Debug]        Searching for tablet 'Genius G-Pen 560'
Dec 22 12:00:47 knoyan.local otd-daemon[4024]: [Detect:Debug]        Searching for tablet 'Genius i405x'
Dec 22 12:00:47 knoyan.local otd-daemon[4024]: [Detect:Debug]        Searching for tablet 'Genius i608x'
Dec 22 12:00:47 knoyan.local otd-daemon[4024]: [Detect:Debug]        Searching for tablet 'Huion 420'
Dec 22 12:00:47 knoyan.local otd-daemon[4024]: [Detect:Debug]        Searching for tablet 'Huion G10T'
...
Dec 22 12:00:47 knoyan.local otd-daemon[4024]: [Detect:Info]        No tablets were detected.
```

lsusb

`Bus 001 Device 014: ID 0458:501a KYE Systems Corp. (Mouse Systems) MousePen i608X`

### What have you tried?

- I tried restart my laptop still not work
- Try Reinstall still not work
- Disable and enable OTD still not work

### Diagnostic Information

[2024-12-22 06_36_48Z.log.json](https://github.com/user-attachments/files/18220989/2024-12-22.06_36_48Z.log.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-12-22:
The log file is completely useless here because it does not include the devices attached to the system. Please export a diagnostic (help > export diagnostics)

KnoyanMitsu, 2024-12-23:
Here

[Log.json](https://github.com/user-attachments/files/18231633/Log.json)

I cant write on comment because limit character

KnoyanMitsu, 2024-12-23:
Hmm i think my ProductName is MousePen i608X

But Genius i608X and Genius MousePen i608X is same anyway

jamesbt365, 2024-12-23:
If you download the config from #3630 and place it in `~/.local/share/OpenTabletDriver/Configurations` it should work.

> Hmm i think my ProductName is MousePen i608X

OpenTabletDriver does not check this when detecting tablets.

KnoyanMitsu, 2024-12-23:
i place ur config and is work now
but is feel weird if i draw on gimp. is pop up GEGL Operation and my cursor is broken. and i try click anything is force me to drag

my issue:
- weird cursor
- not work pressure (artist mode)

maybe u can see this video
https://www.youtube.com/watch?v=anXW67lOGLs
idk why i cant upload video here

and log

[Log.json](https://github.com/user-attachments/files/18232060/Log.json)

jamesbt365, 2024-12-23:
This looks like you have some conflicting application or driver attempting to move your cursor, but it may also just be gnome handling the tablet incorrectly.

Go to tablets > tablet debugger in OpenTabletDriver and see if you see the teleporting behaviour inside the debugger, if you do see the back and forth teleportation we can investigate further, but if you don't its something on your system causing the trouble.

KnoyanMitsu, 2024-12-24:
hmm i see it on tablet debugger visualizer is corectly (no teleport) and pressure and pen is work too. try on kde same problem

maybe cause this...[image]

idk how disable it.... any idea?

gonX, 2025-01-24:
Fixed in #3630, will come in v0.6.5.1 release.

closed 2025-01-24 by gonX

KnoyanMitsu, 2025-01-25:
Alright thanks

PR #3629 (2024-12-28): Add Linux and MacOS support for the Genius i608x
Fixes #3627.