OpenTabletDriver/OpenTabletDriver #3632: Add support for Wacom DTH227K0B (Cintiq Pro 22)

2meito, 2024-12-23:
### Manufacturer

Wacom

### Product Name

DTH227K0B (Cintiq Pro 22)

### Device Width

476

### Device Height

268

### Pressure Levels

8192

### Additional Information

[image]

### Diagnostic Information

[diagnostics.json](https://github.com/user-attachments/files/18232845/diagnostics.json)

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

jamesbt365, 2024-12-23:
[99-opentabletdriver.rules.txt](https://github.com/user-attachments/files/18232867/99-opentabletdriver.rules.txt)
Place this in `/etc/udev/rules.d` called `99-opentabletdriver.rules` (remove the .txt or it may not work, i only had to put it there because github file restrictions.

Once that is done, run the following commands and then export diagnostics again.

```
sudo udevadm control --reload-rules && sudo udevadm trigger
```

2meito, 2024-12-23:
done:
[new-diagnostics.json](https://github.com/user-attachments/files/18232948/new-diagnostics.json)

2meito, 2024-12-23:
here's a fresh one after a system restart just in case:
[new-new-diagnostics.json](https://github.com/user-attachments/files/18233146/new-new-diagnostics.json)

2meito, 2025-04-03:
Got a second (third) opinion from my windows install:
[windows-diagnostics.json](https://github.com/user-attachments/files/19595678/windows-diagnostics.json)

gonX, 2025-10-06:
Support added in #3858 (backported in #4044) and is available from OpenTabletDriver release v0.6.6.0 onwards.

closed 2025-10-06 by gonX