OpenTabletDriver/OpenTabletDriver #3633: Add support for XP Pen Deco 01 V3

LangeRII, 2024-12-23:
### Manufacturer

XP Pen Deco 01 V3

### Product Name

10" x 6.25"

### Device Width

254mm

### Device Height

158,75mm

### Pressure Levels

16384 Levels

### Lines per Inch (LPI)

5080 LPI

### Additional Information

Hello guys! I've been trying all day to get my XPPen Deco 01 V3 working for gaming. My big issue is that the tablet pressure doesn't allow for a dead zone, even though the driver should be able to disable output for minimal input. Through Kariyu, I learned that with OpenTabletDriver, you can set the pressure point at which the tablet triggers a left click. Unfortunately, I just found out that my tablet isn't supported. And here I am. :D

I would be incredibly grateful if you could integrate this tablet as well! That would be absolutely amazing.

Thank you so much, and Merry Christmas!

### Diagnostic Information

[dia.json](https://github.com/user-attachments/files/18234023/dia.json)

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

madnessman666, 2024-12-27:
I have same issue too

madnessman666, 2024-12-27:
Writed config that allows to use just a pen. [01 V3.json](https://github.com/user-attachments/files/18261593/01.V3.json) Discord post with updates https://discord.com/channels/615607687467761684/1321547016806993960 (on official opentabletdriver discord). Lemme know about issues if any

madnessman666, 2024-12-28:
#3644

jamesbt365, 2025-01-30:
Fixed by #3644

closed 2025-01-30 by jamesbt365