OpenTabletDriver/OpenTabletDriver #3634: Add support for Deco mini7 V2

Jok55r, 2024-12-24:
### Manufacturer

XP-Pen

### Product Name

Deco mini7 V2

### Device Width

260.2

### Device Height

162.2

### Pressure Levels

16384

### Lines per Inch (LPI)

5080

### Additional Information

There is already support for Deco mini7 but it doesn't work on the V2 version, I also tried XP-Pen official drivers AND zadig trying different configurations. It's just that you don't have it in supported devices.

### Diagnostic Information

``` OpenTabletDriver doesn't detect my tablet ```

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

jamesbt365, 2024-12-24:
Diagnostic inforamtion (help > export diagnostics) is mandatory to support tablets, if you are unable to provide such information I will close this issue as your request is impossible.

Jok55r, 2024-12-24:
my bad, didn't know how to do that here it is:
```
{
  "App Version": "OpenTabletDriver v0.6.5.0",
  "Build Date": "12/18/2024",
  "Operating System": {
    "Name": "Win32NT",
    "Version": "10.0.19045.0",
    "Attributes": {
      "ServicePack": ""
    }
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\Joker\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\Joker\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\Joker"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\usb#vid_28bd&pid_0948&mi_02#8&7e390b5&0&0002#{dee824ef-729b-4a0e-9c14-b7117d33a817}",
      "Manufacturer": "UGTABLET",
      "ProductName": "Deco Mini7 V2",
      "SerialNumber": "                    ",
      "FriendlyName": "Deco Mini7 V2",
      "VendorID": 10429,
      "ProductID": 2376,
      "InputReportLength": 12,
      "OutputReportLength": 20,
      "FeatureReportLength": 0,
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "2",
        "HID_REPORTS": "02:FF0A:0001"
      }
    },
    {
      "DevicePath": "\\\\?\\hid#vid_258a&pid_1007&mi_01&col05#8&1b5f1323&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "SINOWEALTH",
      "ProductName": "Wired Gaming Mouse",
      "SerialNumber": "",
      "FriendlyName": "Wired Gaming Mouse",
      "VendorID": 9610,
      "ProductID": 4103,
      "InputReportLength": 0,
      "OutputReportLength": 0,
      "FeatureReportLength": 6,
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "1",
        "HID_REPORTS": ""
      }
    },
    {
      "DevicePath": "\\\\?\\hid#pentablet&col05#1&2d595ca7&2&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "XPPen Technology",
      "ProductName": "XPPenTablet",
      "SerialNumber": "00000000",
      "FriendlyName": "XPPenTablet",
      "VendorID": 255,
      "ProductID": 47820,
      "InputReportLength": 65,
      "OutputReportLength": 65,
      "FeatureReportLength": 0,
      "CanOpen": true,
      "DeviceAttributes": {
        "HID_REPORTS": "40:FF00:0001"
      }
    },
[... 2615 lines cut ...]
      "Group": "IPC",
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2024-12-24T20:53:22.7633242+01:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2024-12-24T20:53:22.9357341+01:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}
```

gonX, 2025-10-06:
Support added in #3739 (backport in #3740) and should be available in OpenTabletDriver v0.6.5.1 onwards.

PR #3739 (2025-01-25): Add support for XP-Pen Deco mini7 V2
Fixes #3634.

Verification: https://discord.com/channels/615607687467761684/1328032996371992589/1332561542696140862
Diagnostics: https://discord.com/channels/615607687467761684/1328032996371992589/1332555038022832159
closed 2025-10-06 by gonX