OpenTabletDriver/OpenTabletDriver #3641: Add support for Huion Kamvas 13 Gen 3

commonkestrel, 2024-12-26:
### Manufacturer

Huion

### Product Name

Kamvas 13 Gen 3

### Device Width

361.21

### Device Height

202.77

### Pressure Levels

16384

### Lines per Inch (LPI)

5080

### Additional Information

The Huion Kamvas 13 is supported, but the recent Gen 3 version is not detected by OTD. More information can be found on Huion's [product page](https://huion.com/products/pen_display/Kamvas/kamvas-13-gen-3.html).

### Diagnostic Information

[otd.json](https://github.com/user-attachments/files/18248025/otd.json)

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

jamesbt365, 2024-12-26:
Fixed by #3637.

commonkestrel, 2024-12-26:
sorry, didnt think to look in pull requests! my mistake

closed 2025-01-19 by gonX