OpenTabletDriver/OpenTabletDriver #3643: OTD does not detect XP-Pen Star 03 on 0.6.5

asrieldashie, 2024-12-26:
### Steps to Reproduce

1. Install otd and enable service
2. Plug in the tablet

### Expected Behavior

Tablet gets detected

### Observed Behavior

Tablet doesn't get detected

### Tablet Device

XP-Pen Star 03 V2 (UC-Logic Tablet 1060N as seen in otd 0.6.4 and lsusb)

### Diagnostic Information

[otd.json](https://github.com/user-attachments/files/18250333/otd.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

asrieldashie, 2024-12-26:
I should mention that reverting to 0.6.4 makes the tablet work again
Unfortunately I couldn't reproduce that again as I reinstalled and upgraded my system and they no longer provides Dotnet 6 which is required by 0.6.4

gonX, 2026-10-04:
This should've be fixed by #4538 (v0.6.7)

closed 2026-10-04 by gonX