OpenTabletDriver/OpenTabletDriver #3661: Add support for Medion Life P82018

Flashkingstyler, 2025-01-06:
### Manufacturer

Medion

### Product Name

P82018

### Device Width

254mm

### Device Height

158,8mm

### Pressure Levels

2048

### Lines per Inch (LPI)

4000

### Diagnostic Information

I would like for this Tablet to be supportet by your Driversoftware :)

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

jamesbt365, 2025-01-06:
A diagnostic file is *required* for tablet support requests, as such:
- Export diagnostics through OpenTabletDriver → Help → Export Diagnostics.
- Upload it here as a `.txt` file.

I will reopen your issue when you supply the mandatory information as without it fulfilling the request is impossible.

Flashkingstyler, 2025-01-06:
Thanks, here ist the exportet File [DiagnosticsMedionP82018.txt](https://github.com/user-attachments/files/18323595/DiagnosticsMedionP82018.txt)