OpenTabletDriver/OpenTabletDriver #3664: Remove ENV{LIBINPUT_IGNORE_DEVICE}="1" on udev rules (linux) for wayland compositor compatibility

thepragandsensdiary, 2025-01-07:
### Steps to Reproduce

Is there a reason why the enviroment variable `ENV{LIBINPUT_IGNORE_DEVICE}="1"` is being added to the udev rules of multiple devices by default?
Ex:
```
$ cat /usr/lib/udev/rules.d/70-opentabletdriver.rules
...
...
KERNEL=="hidraw*", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006d", TAG+="uaccess", TAG+="udev-acl"
SUBSYSTEM=="usb", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006d", TAG+="uaccess", TAG+="udev-acl"
SUBSYSTEM=="input", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006d", ENV{LIBINPUT_IGNORE_DEVICE}="1"
...
...
```
The variable hides the device from wayland compositors / wm such as [niri](https://github.com/YaLTeR/niri). Making the tablet [non functional unless the variable is removed](https://github.com/YaLTeR/niri/issues/944)
A regular user has to get its hands dirty so to speak to make the tablet work. I recall having the same problem on GNOME wayland but don't quote me on that, I don't remember.

Removing it by default seems like the best solution, or maybe it's for X compatibility? Change to feature request if it's not a bug.

* Installed via: AUR, Arch linux.

### Expected Behavior

Tablet working under wayland compositors such as hyprland (untested) and niri (tested)

### Observed Behavior

Tablet not moving the cursor and not clicking inside the wayland compositor

### Tablet Device

Gaomon s620

### Diagnostic Information

"N/A".

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

gonX, 2025-01-08:
Haven't tested this recently, but this at least used to be necessary on some tablets as their HID device (plug and play endpoint) is still reporting to the OS despite the vendor endpoint being in use, and no vendor driver is loaded.

If it's still necessary without any other changes, there is a chance we can do something about it in some other way, e.g. if we can "seize" the tablet (tried in #230 for, IIRC, this reason), or change the initialization we do on the tablet, but again this depends on tablet and the latter isn't a simple fix.

Either way, we need to first have a solution in place that allows us to not need `LIBINPUT_IGNORE_DEVICE=1` without causing any regressions. Usually this would come up as "teleporting cursors".

Also see OpenTabletDriver/opentabletdriver.github.io#196 for documentation, and OpenTabletDriver/OpenTabletDriver-udev#1 for original implementation

thepragandsensdiary, 2025-01-09:
Most of that went over my head lol. Editor's note: 2 hours later I kinda get it now.

Well, at least some info on the FAQ should be necessary, I lost 2 hours trying to solve this. That's fine tho, I plan to be a dev. But for the average user, that's not so bueno.

Steamvoker, 2026-03-12:
I just want to add that this rule completely blocked touch capabilities of my Huion Kamvas Pro 19.

I was diagnosing why the touchscreen was not working. I have checked `cat /dev/input/eventN` and it seemed to respond to touch events, but the device was hidden from the `libinput` by the udev rule. Touch is working after setting the rule to 0.