OpenTabletDriver/OpenTabletDriver #3668: Eto special directories ("my documents" et.al.) sometimes return null

gonX, 2025-01-08:
There have been reports of some users being unable to upload diagnostics as well as exporting or importing profiles - according to the stack trace I've narrowed it down to be because of the `Eto.EtoEnvironment.GetFolderPath(Eto.EtoSpecialFolder.Documents)` call can return null.

This is relatively important to fix because it stops users from providing diagnostics.

I'm unable to reproduce this, but I've observed both Windows and Linux users getting this.

---

Stack trace for ExportDiagnostics:
```
System.UriFormatException: Invalid URI: The URI is empty.
   at System.Uri.CreateThis(String uri, Boolean dontEscape, UriKind uriKind, UriCreationOptions& creationOptions)
   at System.Uri..ctor(String uriString)
   at OpenTabletDriver.UX.MainForm.ExportDiagnostics() in /build/source/OpenTabletDriver.UX/MainForm.cs:line 644
```
which corresponds to the following code block:
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/ccb32a5b7208a6ced9b84dae12b8e7de20aa2ab5/OpenTabletDriver.UX/MainForm.cs#L644-L652

---

Stack trace for LoadSettingsDialog:
```
System.UriFormatException: Invalid URI: The URI is empty.
   at System.Uri.CreateThis(String uri, Boolean dontEscape, UriKind uriKind, UriCreationOptions& creationOptions)
   at System.Uri..ctor(String uriString)
   at OpenTabletDriver.UX.MainForm.LoadSettingsDialog() in /home/evest/work/opentabletdriver/src/OpenTabletDriver-0.6.5.0/OpenTabletDriver.UX/MainForm.cs:line 453
   at OpenTabletDriver.UX.MainForm.<ConstructMenu>b__14_3(Object sender, EventArgs e) in /home/evest/work/opentabletdriver/src/OpenTabletDriver-0.6.5.0/OpenTabletDriver.UX/MainForm.cs:line 218
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Eto.GtkSharp.Forms.GtkSynchronizationContext.<>c__DisplayClass0_0.<Post>b__0()
   at GLib.Idle.IdleProxy.Handler()
```
corresponding to:
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/ccb32a5b7208a6ced9b84dae12b8e7de20aa2ab5/OpenTabletDriver.UX/MainForm.cs#L453-L461

---

Stack trace for PromptInstallPlugin:
```
System.UriFormatException: Invalid URI: The URI is empty.
   at System.Uri.CreateThis(String uri, Boolean dontEscape, UriKind uriKind, UriCreationOptions& creationOptions)
   at System.Uri..ctor(String uriString)
   at OpenTabletDriver.UX.Windows.Plugins.PluginManagerWindow.PromptInstallPlugin(Object sender, EventArgs e) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Windows\Plugins\PluginManagerWindow.cs:line 173
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
```

code block:
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/ccb32a5b7208a6ced9b84dae12b8e7de20aa2ab5/OpenTabletDriver.UX/Windows/Plugins/PluginManagerWindow.cs#L173-L182

`git blame` gives me 7c2764a which is from #1700

Eto source code basically abstracts the Document request to .NET's GetFolderPath:
https://github.com/picoe/Eto/blob/e8796200abc1a8eb2bc9f64eeab82a1880def8d7/src/Eto.Wpf/EtoEnvironmentHandler.cs

The solution is likely to use construct the Uri in another way, for example Uri has a `TryCreate()` method.

Kuuuube, 2025-01-08:
Likely solution is to do the same thing we do for appdata and config folders with `GetExistingPath` before creating the Uri

https://github.com/OpenTabletDriver/OpenTabletDriver/blob/7ec0f7ff69379b625eeed4a8527d3983b796e228/OpenTabletDriver.Desktop/Interop/AppInfo/LinuxAppInfo.cs

Mrcubix, 2025-01-08:
It is not null.
It's an empty string, which can either be defaulted to another call or var.

gonX, 2025-01-08:
> Likely solution is to do the same thing we do for appdata and config folders with `GetExistingPath` before creating the Uri
> 
> https://github.com/OpenTabletDriver/OpenTabletDriver/blob/7ec0f7ff69379b625eeed4a8527d3983b796e228/OpenTabletDriver.Desktop/Interop/AppInfo/LinuxAppInfo.cs

We don't really want to create the folders that don't exist.

Also the issue is mostly that the string is empty which confuses the Uri constructor, so we wouldn't be able to determine which folder to create. So the fix is to fall back to home (`~`), or don't set `Directory` for the file dialog.

gonX, 2025-01-24:
Will be fixed in v0.6.5.1 with #3670 and unlikely to be a problem in 0.7.x (given it moves away from Eto), closing.

PR #3670 (2025-01-24): [06x] UX: Rewrite FileDialog directory checks
Fixes #3668 for 0.6.x

Previously, we assumed that `Eto.EtoEnvironment.GetFolderPath` would never return an empty string. It is however apparent that this is actually the case on some systems, seen on both Windows and Linux.

Constructing URI's with an empty string throws an exception, which this commit attempts to avoid. It was not an option to simply wrap the `Uri` constructor, as `Directory` in `SaveFileDialog`/`OpenFileDialog` must only be set if it is a valid Uri, so instead the `SaveFileDialog`/`OpenFileDialog` Eto constructor has been moved  [... cut]
closed 2025-01-24 by gonX