OpenTabletDriver/OpenTabletDriver #3669: Support Request: gaomon m10k not detected

omegaSomeone, 2025-01-08:
### What do you need help with?

I recently got into drawing again; problem is, last time i used my drawing tablet, i was on windows, and now, on Ubuntu, it is not detected by OpenTabletDriver! the console tells me just that: "No tablets were detected."

### What have you tried?

I tried checking if it was plugged in correctly with `lsusb`, and it is detected there. So, I restarted my computer, to no avail. 

### Diagnostic Information

[diagnostics.json](https://github.com/user-attachments/files/18353454/diagnostics.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

jamesbt365, 2025-01-08:
Your tablet isn't visible in the diagnostic file, this is probably because you have another driver or something else blocking access of the tablet to OpenTabletDriver.

omegaSomeone, 2025-01-08:
Thanks! i'll try and see what may block the access, if i find anything, i'll let you know!

omegaSomeone, 2025-01-08:
...And I had forgotten to uninstall Digimend when I saw it didn't work. Uninstalled it, restarted and now it works. Case closed!

closed 2025-01-08 by omegaSomeone