OpenTabletDriver/OpenTabletDriver #3671: Fatal crash upon loading an exported settings file if file references plugin that is not installed

Pranav-Badrinathan, 2025-01-10:
### Steps to Reproduce

I tried loading an exported settings file from a different computer, however that caused OpenTabletDriver to crash very rudely (without an explanation for the crash, with a NullReferenceException). After reading the settings file, I realized this was because the WindowsInk plugin was not installed on this fresh install of  OpenTabletDriver

1. Open the GUI, and install a plugin (like WinowsInk).
2. Use the plugin somewhere in a tablet's settings (like selecting "Windows Ink Absolute Mode" as output mode)
3. Save, and then export the current settings.
4. In a fresh copy of OpenTabletDriver, try to load the settings exported in the previous step. The program should crash at this step.

### Expected Behavior

The program should handle failure gracefully. Attempt to load the settings. If we get an error, abort the loading, revert to previous settings, and communicate this with the user with a modal or something. Ideally, provide the exact line in the settings file that caused this issue. In my case, it was the lines:
```json
      "OutputMode": {
        "Path": "VoiDPlugins.OutputMode.WinInkAbsoluteMode",
```

This clued me into the issue, which I fixed by simply installing the WindowsInk plugin and *then* loading my settings.

### Observed Behavior

Program throws and displays a NullReferenceException, loses connection with the Daemon, and crashes.

### Tablet Device

Huion 1060P

### Diagnostic Information

Not exactly the right place for this, but hope it is helpful. I have explained the issue above, but if you need this, you'll have it.
```
System.NullReferenceException: Object reference not set to an instance of an object.
   at OpenTabletDriver.UX.Controls.Output.OutputModeEditor.UpdateOutputMode(PluginSettingStore store) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\Output\OutputModeEditor.cs:line 123
   at OpenTabletDriver.UX.Controls.Output.OutputModeEditor.<.ctor>b__0_5(Object sender, EventArgs e) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\Output\OutputModeEditor.cs:line 41
   at Eto.PropertyStore.TriggerEvent[T](Object key, Object sender, T args)
   at Eto.Forms.ListControl.OnSelectedValueChanged(EventArgs e)
   at Eto.Forms.ListControl.OnSelectedIndexChanged(EventArgs e)
   at Eto.Forms.ListControl.Callback.OnSelectedIndexChanged(ListControl widget, EventArgs e)
   at Eto.Wpf.Forms.Controls.DropDownHandler`3.<.ctor>b__1_0(Object sender, SelectionChangedEventArgs e)
   at System.Windows.RoutedEventArgs.InvokeHandler(Delegate handler, Object target)
   at System.Windows.EventRoute.InvokeHandlersImpl(Object source, RoutedEventArgs args, Boolean reRaised)
   at System.Windows.UIElement.RaiseEventImpl(DependencyObject sender, RoutedEventArgs args)
   at System.Windows.Controls.ComboBox.OnSelectionChanged(SelectionChangedEventArgs e)
   at Eto.Wpf.Forms.Controls.EtoComboBox.OnSelectionChanged(SelectionChangedEventArgs e)
   at System.Windows.Controls.Primitives.Selector.SelectionChanger.SelectJustThisItem(ItemInfo info, Boolean assumeInItemsCollection)
   at System.Windows.FrameworkElement.OnPropertyChanged(DependencyPropertyChangedEventArgs e)
   at System.Windows.DependencyObject.NotifyPropertyChange(DependencyPropertyChangedEventArgs args)
   at System.Windows.DependencyObject.UpdateEffectiveValue(EntryIndex entryIndex, DependencyProperty dp, PropertyMetadata metadata, EffectiveValueEntry oldEntry, EffectiveValueEntry& newEntry, Boolean coerceWithDeferredReference, Boolean coerceWithCurrentValue, OperationType operationType)
   at System.Windows.DependencyObject.SetValueCommon(DependencyProperty dp, Object value, PropertyMetadata metadata, Boolean coerceWithDeferredReference, Boolean coerceWithCurrentValue, OperationType operationType, Boolean isInternal)
   at System.Windows.DependencyObject.SetValue(DependencyProperty dp, Object value)
   at Eto.Wpf.Forms.Controls.DropDownHandler`3.set_SelectedIndex(Int32 value)
   at Eto.Forms.ListControl.set_SelectedIndex(Int32 value)
   at Eto.Forms.ListControl.set_SelectedValue(Object value)
   at OpenTabletDriver.UX.Controls.Generic.DropDown`1.<>c.<get_SelectedItemBinding>b__4_1(DropDown`1 c, T v) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\Generic\DropDown.cs:line 20
   at Eto.Forms.IndirectBinding`1.SetValue(Object dataItem, T value)
   at Eto.Forms.ObjectBinding`2.set_DataValue(TValue value)
   at Eto.Forms.BindableBinding`2.<>c__DisplayClass7_0`1.<Convert>b__1(T c, TNewValue v)
   at Eto.Forms.IndirectBinding`1.SetValue(Object dataItem, T value)
   at Eto.Forms.ObjectBinding`2.set_DataValue(TValue value)
   at Eto.Forms.DualBinding`1.SetDestination()
   at Eto.Forms.DualBinding`1.HandleSourceChanged(Object sender, EventArgs e)
   at Eto.Forms.DelegateBinding`1.HandleChangedEvent(Object sender, EventArgs e)
   at Eto.Forms.DirectBinding`1.<>c__DisplayClass14_0`1.<Child>g__valueChanged|0(Object sender, EventArgs e)
   at OpenTabletDriver.UX.Controls.Output.OutputModeEditor.OnProfileChanged() in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\Output\OutputModeEditor.cs:line 70
   at OpenTabletDriver.UX.Controls.Output.OutputModeEditor.set_Profile(Profile value) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\Output\OutputModeEditor.cs:line 61
   at OpenTabletDriver.UX.Controls.Output.OutputModeEditor.<>c.<get_ProfileBinding>b__11_1(OutputModeEditor c, Profile v) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\Output\OutputModeEditor.cs:line 81
   at Eto.Forms.IndirectBinding`1.SetValue(Object dataItem, T value)
   at Eto.Forms.ObjectBinding`2.set_DataValue(TValue value)
   at Eto.Forms.DualBinding`1.SetDestination()
   at Eto.Forms.DualBinding`1.HandleSourceChanged(Object sender, EventArgs e)
   at OpenTabletDriver.UX.Controls.ControlPanel.<OnProfileChanged>b__17_0() in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\ControlPanel.cs:line 116
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
```

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

Pranav-Badrinathan, 2025-01-10:
I suppose this could be a feature/enhancement request instead of a bug report, but I think it is a pretty major issue with the program. Idk the codebase, so I obviously can't comment on how this can be done, but I know we would all appreciate a crash free experience :).

And contributors and maintainers, thank you for all your work on this wonderful project. 😄

Mrcubix, 2025-01-10:
Able to replicate it with the current release, don't think it has been fixed in the meantime

Mrcubix, 2025-01-10:
Will give fixing it a try, should be fairly simple provided nothing else breaks (famous last words).
let's see how cooked i am after this day of work

Mrcubix, 2025-01-10:
This is the throw exception btw

```log
Exception has occurred: CLR/System.NullReferenceException
An exception of type 'System.NullReferenceException' occurred in OpenTabletDriver.UX.dll but was not handled in user code: 'Object reference not set to an instance of an object.'
   at OpenTabletDriver.UX.Controls.Output.OutputModeEditor.UpdateOutputMode(PluginSettingStore store) in C:\Users\tinta\Documents\Repo\C#\OTD\Source\OpenTabletDriver-0.6.4.0\OpenTabletDriver.UX\Controls\Output\OutputModeEditor.cs:line 124
   at OpenTabletDriver.UX.Controls.Output.OutputModeEditor.<.ctor>b__0_5(Object sender, EventArgs e) in C:\Users\tinta\Documents\Repo\C#\OTD\Source\OpenTabletDriver-0.6.4.0\OpenTabletDriver.UX\Controls\Output\OutputModeEditor.cs:line 41
   at Eto.PropertyStore.TriggerEvent[T](Object key, Object sender, T args)
   at Eto.Forms.ListControl.OnSelectedValueChanged(EventArgs e)
   at Eto.Forms.ListControl.OnSelectedIndexChanged(EventArgs e)
   at Eto.Forms.ListControl.Callback.OnSelectedIndexChanged(ListControl widget, EventArgs e)
   at Eto.Wpf.Forms.Controls.DropDownHandler`3.<.ctor>b__1_0(Object sender, SelectionChangedEventArgs e)
   at System.Windows.RoutedEventArgs.InvokeHandler(Delegate handler, Object target)
   at System.Windows.EventRoute.InvokeHandlersImpl(Object source, RoutedEventArgs args, Boolean reRaised)
   at System.Windows.UIElement.RaiseEventImpl(DependencyObject sender, RoutedEventArgs args)
   at System.Windows.Controls.ComboBox.OnSelectionChanged(SelectionChangedEventArgs e)
   at Eto.Wpf.Forms.Controls.EtoComboBox.OnSelectionChanged(SelectionChangedEventArgs e)
   at System.Windows.Controls.Primitives.Selector.SelectionChanger.SelectJustThisItem(ItemInfo info, Boolean assumeInItemsCollection)
   at System.Windows.FrameworkElement.OnPropertyChanged(DependencyPropertyChangedEventArgs e)
   at System.Windows.DependencyObject.NotifyPropertyChange(DependencyPropertyChangedEventArgs args)
   at System.Windows.DependencyObject.UpdateEffectiveValue(EntryIndex entryIndex, DependencyProperty dp, PropertyMetadata metadata, EffectiveValueEntry oldEntry, EffectiveValueEntry& newEntry, Boolean coerceWithDeferredReference, Boolean coerceWithCurrentValue, OperationType operationType)
   at System.Windows.DependencyObject.SetValueCommon(DependencyProperty dp, Object value, PropertyMetadata metadata, Boolean coerceWithDeferredReference, Boolean coerceWithCurrentValue, OperationType operationType, Boolean isInternal)
   at System.Windows.DependencyObject.SetValue(DependencyProperty dp, Object value)
   at Eto.Wpf.Forms.Controls.DropDownHandler`3.set_SelectedIndex(Int32 value)
   at Eto.Forms.ListControl.set_SelectedIndex(Int32 value)
   at Eto.Forms.ListControl.set_SelectedValue(Object value)
   at OpenTabletDriver.UX.Controls.Generic.DropDown`1.<>c.<get_SelectedItemBinding>b__4_1(DropDown`1 c, T v) in C:\Users\tinta\Documents\Repo\C#\OTD\Source\OpenTabletDriver-0.6.4.0\OpenTabletDriver.UX\Controls\Generic\DropDown.cs:line 20
   at Eto.Forms.IndirectBinding`1.SetValue(Object dataItem, T value)
   at Eto.Forms.ObjectBinding`2.set_DataValue(TValue value)
   at Eto.Forms.BindableBinding`2.<>c__DisplayClass7_0`1.<Convert>b__1(T c, TNewValue v)
   at Eto.Forms.IndirectBinding`1.SetValue(Object dataItem, T value)
   at Eto.Forms.ObjectBinding`2.set_DataValue(TValue value)
   at Eto.Forms.DualBinding`1.SetDestination()
   at Eto.Forms.DirectBinding`1.<>c__DisplayClass14_0`1.<Child>g__valueChanged|0(Object sender, EventArgs e)
   at OpenTabletDriver.UX.Controls.Output.OutputModeEditor.OnProfileChanged() in C:\Users\tinta\Documents\Repo\C#\OTD\Source\OpenTabletDriver-0.6.4.0\OpenTabletDriver.UX\Controls\Output\OutputModeEditor.cs:line 70
   at OpenTabletDriver.UX.Controls.Output.OutputModeEditor.set_Profile(Profile value) in C:\Users\tinta\Documents\Repo\C#\OTD\Source\OpenTabletDriver-0.6.4.0\OpenTabletDriver.UX\Controls\Output\OutputModeEditor.cs:line 61
   at OpenTabletDriver.UX.Controls.Output.OutputModeEditor.<>c.<get_ProfileBinding>b__11_1(OutputModeEditor c, Profile v) in C:\Users\tinta\Documents\Repo\C#\OTD\Source\OpenTabletDriver-0.6.4.0\OpenTabletDriver.UX\Controls\Output\OutputModeEditor.cs:line 81
   at Eto.Forms.IndirectBinding`1.SetValue(Object dataItem, T value)
   at Eto.Forms.ObjectBinding`2.set_DataValue(TValue value)
   at Eto.Forms.DualBinding`1.SetDestination()
   at OpenTabletDriver.UX.Controls.ControlPanel.<<OnProfileChanged>b__17_0>d.MoveNext() in C:\Users\tinta\Documents\Repo\C#\OTD\Source\OpenTabletDriver-0.6.4.0\OpenTabletDriver.UX\Controls\ControlPanel.cs:line 116
```

Mrcubix, 2025-01-11:
PR was merged, you can either wait for 0.6.5.1 or use the action build, or build the 0.6.x branch.

Action Build : https://github.com/OpenTabletDriver/OpenTabletDriver/actions/runs/12717895960

(Scroll down to download the artifact for your platform)

gonX, 2025-01-16:
Should be fixed in #3672 as instructed by Mrcubix above. Please reopen this issue it isn't fixed.

closed 2025-01-16 by gonX

Pranav-Badrinathan, 2025-01-23:
Oh, I didn't get notifications at all from this thread for some reason.
> Should be fixed in https://github.com/OpenTabletDriver/OpenTabletDriver/pull/3672 as instructed by Mrcubix above. Please reopen this issue it isn't fixed.

I'll give it a test and let you know. Thanks for the swift action.

Pranav-Badrinathan, 2025-01-23:
Just tested it on my end, and it works. The app just deselects the output mode instead of crashing. Many thanks for the swift response and fix.

PR #3672 (2025-01-10): [0.6.x] Fix NullReferenceException when settings with missing output mode area are loaded
Fixes #3671
Also fixes the same issue with Profiles (They are just Settings files)

Needs testing.

Steps to reproduce :

1. Duplicate your current settings file
2. Write anything that isn't a valid output mode in `Path` for the active output mode
3. Try loading that setting file in.