OpenTabletDriver/OpenTabletDriver #3677: Support Request: OTD only detects 6 (also not working) out of 8 aux of XP PEN Deco 01 v1

pc-v2, 2025-01-13:
### What do you need help with?

OTD only shows 6 out of 8 aux button and all button does not work

### What have you tried?

Reinstall OTD

additional:
device string reader
```
Index 1: UGTABLET
Index 2: TABLET G5 9x6
Index 3: F60 _UG_VDC01_q1109
Index 4: F60 _UG_VDC01_q1109
Index 5: UGEE
Index 6: DECO 01
Index 7: F60 _UG_VDC01_q1109
Index 8: F60 _UG_VDC01_q1109
Index 9: F60 _UG_VDC01_q1109
Index 10: F60 _UG_VDC01_q1109
Index 11: F60 _UG_VDC01_q1109
Index 12: F60 _UG_VDC01_q1109
Index 13: F60 _UG_VDC01_q1109
Index 14: F60 _UG_VDC01_q1109
Index 15: F60 _UG_VDC01_q1109
Index 16: F60 _UG_VDC01_q1109
Index 17: F60 _UG_VDC01_q1109
Index 18: F60 _UG_VDC01_q1109
Index 19: F60 _UG_VDC01_q1109
Index 20: F60 _UG_VDC01_q1109
Index 21: F60 _UG_VDC01_q1109
Index 22: F60 _UG_VDC01_q1109
Index 23: Test
Index 24: Hi-Gain
Index 25: Lo-Gain
Index 26: 
Index 27: 
Index 28: 
Index 29: 
Index 30: F60 _UG_VDC01_q1109
Index 31: F60 _UG_VDC01_q1109
Index 32: F60 _UG_VDC01_q1109
Index 33: F60 _UG_VDC01_q1109
Index 34: F60 _UG_VDC01_q1109
Index 35: F60 _UG_VDC01_q1109
Index 36: F60 _UG_VDC01_q1109
Index 37: F60 _UG_VDC01_q1109
Index 38: F60 _UG_VDC01_q1109
Index 39: F60 _UG_VDC01_q1109
Index 40: F60 _UG_VDC01_q1109
Index 41: F60 _UG_VDC01_q1109
Index 42: F60 _UG_VDC01_q1109
Index 43: F60 _UG_VDC01_q1109
Index 44: F60 _UG_VDC01_q1109
Index 45: F60 _UG_VDC01_q1109
Index 46: F60 _UG_VDC01_q1109
Index 47: F60 _UG_VDC01_q1109
Index 48: F60 _UG_VDC01_q1109
Index 49: F60 _UG_VDC01_q1109
Index 50: F60 _UG_VDC01_q1109
Index 51: F60 _UG_VDC01_q1109
Index 52: F60 _UG_VDC01_q1109
Index 53: F60 _UG_VDC01_q1109
Index 54: F60 _UG_VDC01_q1109
Index 55: F60 _UG_VDC01_q1109
Index 56: F60 _UG_VDC01_q1109
Index 57: F60 _UG_VDC01_q1109
Index 58: F60 _UG_VDC01_q1109
Index 59: F60 _UG_VDC01_q1109
[... 936 lines cut ...]
      "Message": "Tablet debugging is enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2025-01-13T09:21:10.9715461+07:00",
      "Group": "Tablet",
      "Message": "Tablet debugging is disabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    }
  ]
}

```

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

jamesbt365, 2025-01-13:
Well what did you bind the buttons to? you have trimmed the diagnostic, so i'm not 100% sure what you have tried already.

pc-v2, 2025-01-13:
> Well what did you bind the buttons to? you have trimmed the diagnostic, so i'm not 100% sure what you have tried already.

trimmed it because github character limit, should I instead upload the diagnostic? 
 
[image]

[image]

[image]

jamesbt365, 2025-01-13:
> trimmed it because github character limit, should I instead upload the diagnostic?

It was preferred to begin with, means I don't have to scroll for 10 seconds to access information at the top of the issue and it usually means it hasn't been tampered with either.

Tablets > tablet debugger, record your screen when you press through all the buttons (or do a data recording in OTD instead) and upload the video (or data recording from  %localappdata%\OpenTabletDriver) here.

pc-v2, 2025-01-13:
[tablet-data.txt](https://github.com/user-attachments/files/18392441/tablet-data.txt)
it only detects pen buttons

https://github.com/user-attachments/assets/617607d1-f556-4ec4-ad93-4d22b2d14566

jamesbt365, 2025-01-13:
Do another recording but do not put your pen on the tablets, it just dirties the data sample as I do not need to see the pen buttons working. I already knew they did.

Just go through the tablet buttons on the tablets then stop recording, I only need the debugger.

pc-v2, 2025-01-13:
> Do another recording but do not put your pen on the tablets, it just dirties the data sample as I do not need to see the pen buttons working. I already knew they did.
> 
> Just go through the tablet buttons on the tablets then stop recording, I only need the debugger.

it's empty

[tablet-data.txt](https://github.com/user-attachments/files/18396339/tablet-data.txt)

I already clicked all button

https://github.com/user-attachments/assets/233ccc41-8e05-4b30-9aa8-7e440c498070

jamesbt365, 2025-01-13:
Is there a physical switch on your tablet to toggle the auxiliary buttons? Look for one

pc-v2, 2025-01-13:
sadly no, this model is outdated and the latest official driver is buggy in windows (XPPenWin_3.4.14.240603)

closed 2025-08-27 by gonX