OpenTabletDriver/OpenTabletDriver #3692: CHT-661 auxiliary buttons do not work

diniamo, 2025-01-19:
### Steps to Reproduce

1. Own a CHT-661
2. Install OTD
3. Plug it in
4. Bind the auxiliary buttons
5. Press them
6. Nothing

### Expected Behavior

The buttons register and press the set keys correctly.

### Observed Behavior

Nothing happens. The light on the tablet does flash when I press the buttons, so they aren't broken, and I assume they don't need to be toggled anywhere else either. The buttons on the pen function correctly. I've tried on Windows as well, and the same issue happens.

### Tablet Device

CHT-661

### Diagnostic Information

File because GitHub said that it's too many characters otherwise: [diagnostics.json](https://github.com/user-attachments/files/18467546/diagnostics.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

gonX, 2025-01-19:
Can confirm that the CTH-661 parser does not implement aux keys:

Parser definition:
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/d3ff6b720904170db93ff67aaf53f6c7a27ad102/OpenTabletDriver.Configurations/Configurations/Wacom/CTH-661.json#L25

Parser code:
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/d3ff6b720904170db93ff67aaf53f6c7a27ad102/OpenTabletDriver.Configurations/Parsers/Wacom/Intuos/IntuosReportParser.cs#L7-L22

no `IAuxReport` inheritance on this struct:
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/d3ff6b720904170db93ff67aaf53f6c7a27ad102/OpenTabletDriver.Configurations/Parsers/Wacom/Intuos/IntuosTabletReport.cs#L7

@diniamo please make a _tablet data recording_ (OpenTabletDriver GUI -> Tablets -> Tablet Debugger -> Enable Data Recording) in which you are pressing the aux keys from top left to bottom right. then find `tablet-data.txt` in OTD's [application data directory](https://opentabletdriver.net/Wiki/FAQ/General#appdata) and upload it here.
Make sure to stop the recording once you've clicked all the buttons, and do not bring the tablet pen into proximity while recording this.

diniamo, 2025-01-19:
Erm, the debugger doesn't seem to record data from the aux keys.

jamesbt365, 2025-01-24:
Fixed in #3694.

PR #3694 (2025-01-22): [06x] CTH-*61: Add auxiliary key support
This adds a new aux report parser, but the report matches the offset on IntuosV2AuxReport, so we reuse that

Fixes #3692

[Discord verification](https://discord.com/channels/615607687467761684/723399565780189234/1330557683144851468)

Example output from a CTH-661:
```
{ 02 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 }
{ 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 }
{ 02 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 }
{ 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 }
{ 02 04 00 00 00 00 00 00 00 00 00 [... cut]
closed 2025-01-24 by jamesbt365