OpenTabletDriver/OpenTabletDriver #3738: Add support for XP-PEN Star 02

MellowVellow, 2025-01-25:
### Manufacturer

XP-PEN

### Product Name

Star 02

### Diagnostic Information

[STAR02.json](https://github.com/user-attachments/files/18544800/STAR02.json)

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

jamesbt365, 2025-01-25:
It is not possible to proceed unless you do this.

[70-opentabletdriver.rules.txt](https://github.com/user-attachments/files/18544802/70-opentabletdriver.rules.txt)

Move the above file to `/etc/udev/rules.d/70-opentabletdriver.rules` (note the lack of .txt) then run the following commands:
```
sudo udevadm control --reload-rules && sudo udevadm trigger
```

MellowVellow, 2025-01-25:
I'm not sure what that does, it doesn't provide any console output, and KDE Plasma can't detect UC-Logic PEN TABLET Stylus (Star 02) after doing your instructions

jamesbt365, 2025-01-25:
> and KDE Plasma can't detect UC-Logic PEN TABLET Stylus (Star 02) after doing your instructions

Its not supposed to, this is a good sign. Export diagnostics again.

> the website says related to the star model 03, "Older variants may use the same configuration as the UC-Logic 1060N."
but there doesn't seem to be any documentation of how to do this

It would do it automatically and star 02 != star 03.

MellowVellow, 2025-01-25:
Alright, here is the new diagnostics

[STAR02.json](https://github.com/user-attachments/files/18546528/STAR02.json)

jamesbt365, 2025-01-25:
Create `~/.local/share/OpenTabletDriver/Configurations` and place [this](https://github.com/user-attachments/files/18546673/star02.json) file inside there, then replug the tablet.

This file does not mean support is complete, so do not close this issue. I need you to verify some functionality first.

Tablets > tablet debugger, move your pen to the bottom right of the tablet and get the highest x and y coordinates you can, I need these to map to the tablet correctly.

I also need the max pressure, if the pen buttons go to true when you press them and if the tablet buttons appear when you press them in the tablet report (take your pen off the tablet for this step, it will be easier to see whats going on).

MellowVellow, 2025-01-26:
After matching the vendor and product Id i got it to detect the tablet, 
Max position:  [8000,8000] (pen pressure cant be detected ~7995)
Max pen pressure: 2047
Both pen buttons work
Tablet buttons don't show up in the tablet report but is detected/reported

[star02.json](https://github.com/user-attachments/files/18548592/star02.json)

jamesbt365, 2025-01-26:
Why did you modify the file and send it back? Parameters like the Interface attribute were intentionally placed to make sure that conflicts wouldn't occur, please use the file I sent you with no modifications.

I will do the modifications that are fit when I know what does and doesn't work. Please change the config back to the one I originally gave you and **replug the tablet**, testing again for behaviour.

MellowVellow, 2025-01-26:
yeah, works just the same

jamesbt365, 2025-01-26:
Export diagnostics

MellowVellow, 2025-01-26:
[STAR02.json](https://github.com/user-attachments/files/18548657/STAR02.json)

jamesbt365, 2025-01-26:
Does changing the interface attribute to a different value (1 or 2) result in different behaviour? (do keep in mind you have to replug)

MellowVellow, 2025-01-26:
both 1 and 2 result in no reports in the debugger

jamesbt365, 2025-01-26:
Try this one, if it appears in the debugger check max x and y: 

[star02.json](https://github.com/user-attachments/files/18548678/star02.json)

MellowVellow, 2025-01-26:
Max position is still [8000,8000]

jamesbt365, 2025-01-26:
[star02.json](https://github.com/user-attachments/files/18548696/star02.json) Try this one (remember, replugs between)

MellowVellow, 2025-01-26:
behaviour is still the same,

jamesbt365, 2025-01-26:
Tablets -> Device String reader

21827
80
leave the bottom one blank, dump all, save that file and send it here

MellowVellow, 2025-01-26:
[dump.txt](https://github.com/user-attachments/files/18548705/dump.txt)

jamesbt365, 2025-01-26:
[star02.json](https://github.com/user-attachments/files/18548713/star02.json)

MellowVellow, 2025-01-26:
max position is now [32000,20000]

jamesbt365, 2025-01-26:
Please check the rest of the stuff like I asked for originally, hopefully auxiliary keys now appear in the debugger.

MellowVellow, 2025-01-26:
tablet buttons still don't respond, and pressing the pen buttons result in pen out of range everytime it is pressed

jamesbt365, 2025-01-26:
Could you either do a screen or a data recording for both the pen and tablet buttons please?

If you're going to use the inbuilt data recorder the file will be `~/.config/OpenTabletDriver/tablet-data.txt`

jamesbt365, 2025-01-26:
Actually, thats probably not neccessary, this has UC-Logic firmware opposed to XP-Pen firmware, this was made when XP-Pen still used that.

Let me know how this goes
[star02.json](https://github.com/user-attachments/files/18548750/star02.json)

MellowVellow, 2025-01-26:
the tablet is fully mapped now, but the pen out of range text behaviour is replaced with this text (tablet buttons, pen buttons)
"{ 07 C0 1B 4E 8A 15 00 00 }, Delta:28.131ms, AuxButtons:[False True False True False False False True True False True False True False False False False False False False], ReportType:UCLogicAuxReport"

jamesbt365, 2025-01-26:
Seems like this needs a new parser for the auxiliary keys to work, could you test with this to check if the pen buttons work at least?

[star02.json](https://github.com/user-attachments/files/18548807/star02.json) 

If they do work, could you do a data recording of you pressing through all 4 auxiliary keys (i know they won't appear, but the raw data will be useful)

MellowVellow, 2025-01-26:
Only the first pen button seems to work, the second will return 0 pen pressure
[tablet-data.txt](https://github.com/user-attachments/files/18548846/tablet-data.txt)

jamesbt365, 2025-01-26:
Not really sure what's going on there as you've thrown a big data file at me without really explaining what you did. When trying to figure out what is and isn't working properly I need a clear idea of what you're doing as a substitute.

Please send separate recordings of you: pressing the first pen button, releasing, then the second one and releasing

Ideally keep movement to a minimum.

And a second recording of **just** pressing the first button, releasing, then second, releasing and so forth. 

Don't touch the pen at all, I need to know what data is mapping to and a dirty sample doesn't make that easy.

MellowVellow, 2025-01-26:
Sure, but I'm not sure if it's because my tablet is old or if the pen has to hover over the tablet to report button presses, 
anyway here's a manually recovered recording, is this clear?

```
#First button
{ 07 80 DB 00 2C 01 00 00 }, Delta:104.745ms, Position:[219,300], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 80 DB 00 2C 01 00 00 }, Delta:0.303ms, Position:[219,300], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 80 DB 00 2C 01 00 00 }, Delta:2.430ms, Position:[219,300], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 82 DB 00 2C 01 00 00 }, Delta:18.659ms, Position:[219,300], Pressure:0, PenButtons:[True False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 82 DB 00 2C 01 00 00 }, Delta:0.330ms, Position:[219,300], Pressure:0, PenButtons:[True False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 82 DB 00 2C 01 00 00 }, Delta:0.290ms, Position:[219,300], Pressure:0, PenButtons:[True False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 82 DB 00 3D 01 00 00 }, Delta:0.286ms, Position:[219,317], Pressure:0, PenButtons:[True False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 82 C9 00 3D 01 00 00 }, Delta:0.283ms, Position:[201,317], Pressure:0, PenButtons:[True False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 82 C9 00 3D 01 00 00 }, Delta:4.010ms, Position:[201,317], Pressure:0, PenButtons:[True False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 82 C9 00 56 01 00 00 }, Delta:4.355ms, Position:[201,342], Pressure:0, PenButtons:[True False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 C0 C9 00 56 01 00 00 }, Delta:3.921ms, Position:[201,342], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 80 35 01 55 01 00 00 }, Delta:600.786ms, Position:[309,341], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 80 35 01 55 01 00 00 }, Delta:0.147ms, Position:[309,341], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 C0 35 01 55 01 00 00 }, Delta:2.927ms, Position:[309,341], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 80 C2 00 56 01 00 00 }, Delta:63.936ms, Position:[194,342], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 80 C2 00 56 01 00 00 }, Delta:20.676ms, Position:[194,342], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport

#Second button
{ 07 80 07 01 E4 01 00 00 }, Delta:3.733ms, Position:[263,484], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 80 07 01 E4 01 00 00 }, Delta:4.064ms, Position:[263,484], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 80 F5 00 CE 01 00 00 }, Delta:19.355ms, Position:[245,462], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 80 F5 00 CE 01 00 00 }, Delta:0.105ms, Position:[245,462], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 80 F5 00 CE 01 00 00 }, Delta:0.074ms, Position:[245,462], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 C0 F5 00 CE 01 00 00 }, Delta:0.081ms, Position:[245,462], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 80 E5 00 AF 01 00 00 }, Delta:12.367ms, Position:[229,431], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 80 E5 00 AF 01 00 00 }, Delta:20.319ms, Position:[229,431], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 84 E5 00 AF 01 00 00 }, Delta:0.102ms, Position:[229,431], Pressure:0, PenButtons:[False True], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 84 E5 00 AF 01 00 00 }, Delta:0.081ms, Position:[229,431], Pressure:0, PenButtons:[False True], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 84 E5 00 AF 01 00 00 }, Delta:0.120ms, Position:[229,431], Pressure:0, PenButtons:[False True], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 84 E5 00 AF 01 00 00 }, Delta:0.066ms, Position:[229,431], Pressure:0, PenButtons:[False True], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 84 E5 00 AF 01 00 00 }, Delta:3.291ms, Position:[229,431], Pressure:0, PenButtons:[False True], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 84 E5 00 AF 01 00 00 }, Delta:3.918ms, Position:[229,431], Pressure:0, PenButtons:[False True], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 84 E5 00 AF 01 00 00 }, Delta:5.137ms, Position:[229,431], Pressure:0, PenButtons:[False True], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 84 E5 00 C3 01 00 00 }, Delta:16.039ms, Position:[229,451], Pressure:0, PenButtons:[False True], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 84 E5 00 C3 01 00 00 }, Delta:0.192ms, Position:[229,451], Pressure:0, PenButtons:[False True], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
[... cut ...]
{ 07 80 6D 01 00 03 00 00 }, Delta:16.549ms, Position:[365,768], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 80 6D 01 00 03 00 00 }, Delta:1.022ms, Position:[365,768], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 80 6D 01 00 03 00 00 }, Delta:0.211ms, Position:[365,768], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 80 6D 01 10 03 00 00 }, Delta:0.079ms, Position:[365,784], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
```

jamesbt365, 2025-01-26:
Pen buttons look like they are reporting fine, though you haven't really said what you have been pressing from the tablet buttons so once again its a bit ambiguous, if you *have* to have the pen on the tablet please don't press any pen buttons while doing the recording.

MellowVellow, 2025-01-26:
Right my bad,
although tablet buttons don't report anything so that's why i forgot.

this has become quite a long thread huh

MellowVellow, 2025-01-26:
The original config couldn't get readings, I had to modify it abit
```
{
  "Name": "XP-Pen Star 02",
  "Specifications": {
    "Digitizer": {
      "Width": 203.2,
      "Height": 127,
      "MaxX": 32000,
      "MaxY": 20000
    },
    "Pen": {
      "MaxPressure": 2047,
      "Buttons": {
        "ButtonCount": 2
      }
    },
    "AuxiliaryButtons": {
      "ButtonCount": 5
    }
  },
  "DigitizerIdentifiers": [
    {
      "VendorID": 21827,
      "ProductID": 80,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "ReportParser": "OpenTabletDriver.Configurations.Parsers.UCLogic.UCLogicReportParser",
      "OutputInitReport": [],
      "DeviceStrings": {
      },
      "InitializationStrings": [
        "100",
        "123"
      ]
    }
  ],
  "Attributes": {
    "libinputoverride": "1",
    "Interface": "0"
  }
}

```

Using the ```OpenTabletDriver.Plugin.Tablet.TabletReportParser```

```
{ 07 E0 01 10 01 00 00 00 }, Delta:3266.410ms, Position:[4097,1], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 E0 01 10 00 00 00 00 }, Delta:213.475ms, Position:[4097,0], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 E0 01 10 FF 00 00 00 }, Delta:1952.835ms, Position:[4097,255], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 E0 01 10 00 00 00 00 }, Delta:175.231ms, Position:[4097,0], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 E0 01 01 01 00 00 00 }, Delta:2504.932ms, Position:[257,1], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 E0 01 01 00 00 00 00 }, Delta:266.211ms, Position:[257,0], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 E0 01 01 02 00 00 00 }, Delta:4404.060ms, Position:[257,2], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 E0 01 01 00 00 00 00 }, Delta:243.785ms, Position:[257,0], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 E0 01 01 04 00 00 00 }, Delta:964.539ms, Position:[257,4], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport
{ 07 E0 01 01 00 00 00 00 }, Delta:135.024ms, Position:[257,0], Pressure:0, PenButtons:[False False], ReportType:OpenTabletDriver.Plugin.Tablet.TabletReport

```
using this config I got a more clear reading 
```
{ 07 E0 01 10 01 00 00 00 }, Delta:4693.362ms, AuxButtons:[True False False False False False False False False False False False False False False False False False False False], ReportType:UCLogicAuxReport
{ 07 E0 01 10 00 00 00 00 }, Delta:331.259ms, AuxButtons:[False False False False False False False False False False False False False False False False False False False False], ReportType:UCLogicAuxReport
{ 07 E0 01 10 FF 00 00 00 }, Delta:1284.466ms, AuxButtons:[True True True True True True True True False False False False False False False False False False False False], ReportType:UCLogicAuxReport
{ 07 E0 01 10 00 00 00 00 }, Delta:314.152ms, AuxButtons:[False False False False False False False False False False False False False False False False False False False False], ReportType:UCLogicAuxReport
{ 07 E0 01 01 01 00 00 00 }, Delta:1106.846ms, AuxButtons:[True False False False False False False False False False False False False False False False False False False False], ReportType:UCLogicAuxReport
{ 07 E0 01 01 00 00 00 00 }, Delta:327.857ms, AuxButtons:[False False False False False False False False False False False False False False False False False False False False], ReportType:UCLogicAuxReport
{ 07 E0 01 01 02 00 00 00 }, Delta:1116.015ms, AuxButtons:[False True False False False False False False False False False False False False False False False False False False], ReportType:UCLogicAuxReport
{ 07 E0 01 01 00 00 00 00 }, Delta:330.357ms, AuxButtons:[False False False False False False False False False False False False False False False False False False False False], ReportType:UCLogicAuxReport
{ 07 E0 01 01 04 00 00 00 }, Delta:621.651ms, AuxButtons:[False False True False False False False False False False False False False False False False False False False False], ReportType:UCLogicAuxReport
{ 07 E0 01 01 00 00 00 00 }, Delta:396.439ms, AuxButtons:[False False False False False False False False False False False False False False False False False False False False], ReportType:UCLogicAuxReport

```
pen buttons, pen pressure & drawing space seems to correct now, although I can't confirm because my pen buttons are barely functional, it's just these tablet buttons bytes are mapped incorrectly with 20 of those bools

jamesbt365, 2025-01-26:
Honestly, we might just have to add it as is, your pen is partially broken and that could be responsible for why the tablet is sending what would be reported as out of range reports. It is just too hard to tell what is and isn't broken.

Given as your tablet for whatever reason requires the pen to be on the tablet to report auxiliary keys, we can't really verify those either but outside of the weird `FF` they seem to work fine.

Can't really get a clean data sample with your broken tablet so its impossible to really improve support further than what we have now, I'll PR this under missing features and hopefully someone in the future can verify its functionality with a working tablet.

If you could export a diagnostic with the current configuration so we can see it working, that would assist in getting this in the driver.

MellowVellow, 2025-01-27:
[STAR02.json](https://github.com/user-attachments/files/18552353/STAR02.json)
another thing I want to mention
[TabletDebugger:Warning]	Touch undefined in tablet configuration - unable to draw touch points

jamesbt365, 2025-01-27:
> [TabletDebugger:Warning] Touch undefined in tablet configuration - unable to draw touch points

Not an issue, your tablet doesn't have touch.

gonX, 2025-10-06:
Support added in #3753 (backport in #3754) - OpenTabletDriver release v0.6.5.1 onwards should support this tablet now.

PR #3753 (2025-02-02): Add support for XP-Pen Star 02
Fixes #3738, all information is within that thread.

Pen and tablet buttons may cause malformed reports, but the user has mentioned that their pen is broken and because of that its too hard to tell what is good data and what is bad data. 

Supporting this now in hopes that somebody can help us confirm in the future if this works.
closed 2025-10-06 by gonX