OpenTabletDriver/OpenTabletDriver #3741: touch input support

Ontario199, 2025-01-25:
### What do you need help with?

is there a plugin to support touch input?

### What have you tried?

 I tried "Touch Gestures" but it didn't help.

### Diagnostic Information

[OpenTablet.json](https://github.com/user-attachments/files/18545200/OpenTablet.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

jamesbt365, 2025-01-25:
Alongside OpenTabletDriver not having touch support natively, and that we dont support plugins here, this is outside the scope of the driver right now and a duplicate of #1664.

However, its not working because you have multiple drivers installed.

closed 2025-01-25 by jamesbt365

Mrcubix, 2025-01-25:
You say you tried Touch Gestures but didn't go to its wiki where i have wrote everything you needed to do to make it work.
That includes additional dependencies, and a first setup guide for binding a new gesture to an action,