OpenTabletDriver/OpenTabletDriver #3748: [06x] Reopening about dialog causes NRE

gonX, 2025-01-25:
How to reproduce:

- Open about dialog (Help -> About, or press F1)
- Close dialog (press the Close button or press Escape)
- Open dialog again
- Receive `NullReferenceException`

Stack trace:
```
System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation.
 ---> System.NullReferenceException: Object reference not set to an instance of an object.
   at Eto.PlatformDetect.GetCallingAssembly()
   at InvokeStub_PlatformDetect.GetCallingAssembly(Object, Object, IntPtr*)
   at System.Reflection.MethodBaseInvoker.InvokeWithNoArgs(Object obj, BindingFlags invokeAttr)
   --- End of inner exception stack trace ---
   at System.Reflection.MethodBaseInvoker.InvokeWithNoArgs(Object obj, BindingFlags invokeAttr)
   at Eto.Forms.AboutDialog..ctor()
   at OpenTabletDriver.UX.App.get_AboutDialog() in /home/gonx/prog/_opentabletdriver/opentabletdriver/OpenTabletDriver.UX/App.cs:line 138
   at OpenTabletDriver.UX.MainForm.<ConstructMenu>b__13_1(Object sender, EventArgs e) in /home/gonx/prog/_opentabletdriver/opentabletdriver/OpenTabletDriver.UX/MainForm.cs:line 212
   at Eto.Forms.Command.OnExecuted(EventArgs e)
   at Eto.Forms.Command.Execute()
   at Eto.Forms.Command.System.Windows.Input.ICommand.Execute(Object parameter)
   at Eto.PropertyStore.CommandWrapper.Command_Execute(Object sender, EventArgs e)
   at Eto.Forms.MenuItem.OnClick(EventArgs e)
   at Eto.Forms.MenuItem.Callback.OnClick(MenuItem widget, EventArgs e)
   at Eto.GtkSharp.Forms.Menu.ButtonMenuItemHandler`2.ButtonMenuItemConnector.HandleActivated(Object sender, EventArgs e)
   at InvokeStub_EventHandler.Invoke(Object, Span`1)
   at System.Reflection.MethodBaseInvoker.InvokeWithFewArgs(Object obj, BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
```

Mrcubix, 2025-01-25:
Can't reproduce on Windows.

gonX, 2025-08-27:
Still happens for me on 0.6.5.1 on Linux

gonX, 2025-09-29:
Fixed in #4103