OpenTabletDriver/OpenTabletDriver #3765: Eraser Tool not reported to libinput on linux

OmegaRogue, 2025-01-29:
### Steps to Reproduce

1. open any tablet tester tool
2. touch the tablet with the eraser tip

[image]

[image]

### Expected Behavior

the eraser tool is reported to libinput

### Observed Behavior

a normal pen tool is reported to libinput. When this is instead done with the opentabletdriver tablet tester, the eraser is correctly detected.

[image]

### Tablet Device

Wacom CTE-430, Wacom ET-0405-U

### Diagnostic Information

[diag.json](https://github.com/user-attachments/files/18588006/diag.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

jamesbt365, 2025-01-29:
Could you test on the latest `0.6.x` branch's commit? Chances are we already fixed it.

OmegaRogue, 2025-01-29:
> Could you test on the latest `0.6.x` branch's commit? Chances are we already fixed it.

on my first test this reported `BTN_TOOL_PEN` for the pen tip and `BTN_TOOL_PEN BTN_TOOL_RUBBER` for the eraser tip, on the second try after looking around the settings both reported no buttons

OmegaRogue, 2025-01-29:
Actually, i noticed i used the master branch instead of the 0.6.x branch, trying again with the 0.6.x branch everything works fine

closed 2025-01-29 by jamesbt365