OpenTabletDriver/OpenTabletDriver #3776: Wacom Intuos started "dragging" with 0.6.5 on MacOS

aapokojo, 2025-01-30:
### Steps to Reproduce

1. Open a drawing software (e.g. Figma, Whimsical)
2. Use a freehand tool (e.g. Marker on FigJam or Marker on Whimsical)
3. Draw a line with the pen on the Wacom
4. Lift the pen

### Expected Behavior

Line is drawn, and stops drawing when the pen is lifted.

### Observed Behavior

Pen keeps on drawing even after lifting the pen from the tablet and hovering over the tablet and stops drawing only when tapping the tablet again with the pen.

Downgrading the driver back to version 0.6.4. solved the issue.
Using MacOS Sonoma 14.5

### Tablet Device

Wacom CTL-490

### Diagnostic Information

N/A

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

gonX, 2025-01-30:
Is this perhaps fixed with #3764 ?

gonX, 2025-01-31:
We now have a release candidate out for the next version (0.6.5.1rc1). Please try that and check whether it fixes your issue: https://github.com/OpenTabletDriver/OpenTabletDriver/actions/runs/13061164724

closed 2025-08-27 by gonX