OpenTabletDriver/OpenTabletDriver #3789: Please add support for XP-Pen Deco Pro MW Gen2

123unix, 2025-02-01:
### Manufacturer

XP-Pen

### Product Name

Deco Pro MW Gen2

### Device Width

283mm - device size, 9in drawing area

### Device Height

227mm - device size, 6in drawing area

### Pressure Levels

16384

### Lines per Inch (LPI)

5080

### Additional Information

XP-Pen Deco Pro **LW** Gen2 is listed as supported, therefore my guess is supporting **MW** would be a minor configuration tweak to adjust for the size.

### Diagnostic Information

[otd-diag-export.json](https://github.com/user-attachments/files/18629216/otd-diag-export.json)

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

jamesbt365, 2025-02-01:
> "HID Devices": [],

There is a misconfiguration somewhere either with permissions or your kernel that makes it impossible for OTD to read any of your USB devices, thus its impossible to support your tablet without amending this first.

123unix, 2025-02-01:
[otd-diag-export-2.json](https://github.com/user-attachments/files/18629713/otd-diag-export-2.json)
Updated diagnostics ouput

jamesbt365, 2025-02-01:
Do you have the XP-Pen driver installed? If you do, it seems to be blocking an interface from being visible.

123unix, 2025-02-01:
Yes, I've installed stock driver after having otd not detecting the tablet.

You know, I really would like to have otd support this tablet, so please let me know how to help or volunteer with a PR, but I also have an interesting observation.

After connecting the tablet via the stock driver with USB the tablet continues working fine with USB unplugged. Officially there is no Bluetooth support for this model. 

How come it works wireless? What do you think? Will it work wireless with OTD?

jamesbt365, 2025-02-01:
>  I really would like to have otd support this tablet

Well in order for OTD to be able to do that you need to get rid of everything that will conflict (other drivers) and export another diagnostic. Otherwise I do not have the information to be able to do that.

> How come it works wireless? What do you think? Will it work wireless with OTD?

Maybe, and we won't know until we see how it acts and if it acts differently.

imdryze, 2025-02-05:
Hello, i've made a new diagnostic without any other drivers installed on my pc and if that can help you 

[diag.json](https://github.com/user-attachments/files/18664719/diag.json)

jensMF, 2025-02-07:
I have a diagnostic, too:

And I created a config file based on the LW and XLW Gen2 files:

jensMF, 2025-02-12:
@jamesbt365 Can you please have a look if there is still any information missing?

Kuuuube, 2025-04-04:
#3860

closed 2025-04-04 by Kuuuube