OpenTabletDriver/OpenTabletDriver #3798: Pen Buttons not working on Fedora 41

KelvinNovais, 2025-02-07:
### Steps to Reproduce

1. Set OTD to "Artist Mode", on the Output page
2. Set `Pen Settings > Pen Buttons > Pen Biding 1` to "Pen Button 1"
3. Set `Pen Settings > Pen Buttons > Pen Biding 2` to "Pen Button 2"
4. Open a software (like Xournal++ or Rnote) and try to use the pen buttons

### Expected Behavior

The pen button should activate the tool configured on the software

### Observed Behavior

When the any pen button is pressed, it _seems_ to activate the left mouse button.

---

### Additional information

I was already experiencing this issue:

-  by building ODT from source, before [v0.6.5.0](https://github.com/OpenTabletDriver/OpenTabletDriver/releases/tag/v0.6.5.0) was released
- when using [v0.6.5.0](https://github.com/OpenTabletDriver/OpenTabletDriver/releases/tag/v0.6.5.0)

This behavior was **not** observed on Fedora 40 with OTD [v0.6.4.0](https://github.com/OpenTabletDriver/OpenTabletDriver/releases/tag/v0.6.4.0)

https://github.com/user-attachments/assets/58611cb7-c6ec-4724-b4cf-7d767dc0f253

### Tablet Device

XP-Pen CT1060

### Diagnostic Information

[diagnostic.json](https://github.com/user-attachments/files/18714441/diagnostic.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

jamesbt365, 2025-02-07:
I highly doubt this is a bug and is much more likely to be a misconfiguration within your desktop environment.

The behaviour of the artist mode buttons are dependent on the settings of said desktop environment and even the applications within it, we just send the proper events and you should be able to see this using `sudo libinput debug-tablet`. 

If the respective buttons pop up there, then its a misconfiguration within your desktop environment.

KelvinNovais, 2025-02-07:
> If the respective buttons pop up there, then its a misconfiguration within your desktop environment.

Yes, you're right! Sorry for opening an issue, but I couldn't figure it out before. If you don't bother pointing me some documentation to fix it, I'll be even more grateful.

closed 2025-02-07 by KelvinNovais

KelvinNovais, 2025-02-11:
See https://discourse.gnome.org/t/graphics-tablet-pen-buttons-are-not-working-correctly-after-upgrading-to-fedora-41/24836/13

KelvinNovais, 2025-04-26:
Just notifying, this external bug was **fixed on Fedora 42**.