OpenTabletDriver/OpenTabletDriver #3799: Support Request: XP-Pen Deco 03

micartey, 2025-02-09:
### What do you need help with?

OpenTabletDriver does not support deco 03 yet

### What have you tried?

Checked the list of [supported devices](https://opentabletdriver.net/Tablets)

### Diagnostic Information

[Diagnostics.json](https://github.com/user-attachments/files/18722923/Diagnostics.json)

### Acknowledgements

- [ ] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [ ] I am on the latest version of OpenTabletDriver.

jamesbt365, 2025-02-09:
> ### What have you tried?
> 
> Checked the list of [supported devices](https://opentabletdriver.net/Tablets)

https://opentabletdriver.net/Tablets?search=deco%2003

What isn't right here is that your udev rules aren't properly loaded, which may be caused from a number of misconfigurations in your nix config, ensure OTD's udev rules are being properly loaded and your tablet will detect.

closed 2025-02-09 by jamesbt365

micartey, 2025-02-09:
Sorry for the useless issue...

1. This was one of the cases where I should have done a reboot... It works now
2. I didn't use the search feature, but `STRG+F`. And there are two groups of `XP-Pen Deco` tablets in the list that have a few other tablets in between, and I didn't go through all the searches but rather scrolled through the highlighted ones in that group... Sorry for being blind 🫠