OpenTabletDriver/OpenTabletDriver #3804: Cannot start the daemon after switching user because `CoreFxPipe_OpenTabletDriver.Daemon` is not removed from `/tmp`

MathisP75, 2025-02-14:
### Steps to Reproduce

1. Enable the daemon as a systemd service and start it.
2. Exit your session and log into a separate user.
3. Launch the daemon. 

### Expected Behavior

The daemon should start.

### Observed Behavior

The daemon does not start, and instead crashes with `[SocketException:Fatal] Address already in use`. This is because there is still a `CoreFxPipe_OpenTabletDriver.Daemon` inside `/tmp`. There is probably something missing to delete this file after a user logs out of his session.

### Tablet Device

XP-Pen Artist Pro 16TP

### Diagnostic Information

[dignostic.json](https://github.com/user-attachments/files/18804054/dignostic.json)

### Acknowledgements

- [ ] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [ ] I am on the latest version of OpenTabletDriver.

CrystalSplitter, 2025-04-14:
Also recently encountered this exact same bug. Thank you for including the bit about `/tmp/CoreFxPipe_OpenTabletDriver.Daemon`, was easy to verify due to it.

CrystalSplitter, 2025-04-14:
(For the record, the workaround for me was to manually force delete these with `sudo rm /tmp/CoreFxPipe_OpenTabletDriver.*`