OpenTabletDriver/OpenTabletDriver #3806: Add support for spacedesk (virtual tablet)

quadrinex, 2025-02-15:
### Manufacturer

datronicsoft

### Product Name

spacedesk

### Additional Information

Spacedesk is an app that allow you to use any device as a secondary screen and have support for stylus (eg. S-pen from samsung) (support windows Ink).

### Diagnostic Information

```json
{
  "App Version": "OpenTabletDriver v0.6.5.1",
  "Build Date": "2/2/2025",
  "Operating System": {
    "Name": "Win32NT",
    "Version": "10.0.26100.0",
    "Attributes": {
      "ServicePack": ""
    }
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\Cyril\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\Cyril\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\Cyril"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_1a2c&pid_9605&mi_01&col06#8&29a910f3&0&0005#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd",
      "Manufacturer": "SEMICO  ",
      "ProductName": "USB Gaming Keyboard ",
      "SerialNumber": "",
      "FriendlyName": "USB Gaming Keyboard ",
      "VendorID": 6700,
      "ProductID": 38405,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "1",
        "HID_REPORTS": "06:0001:0006"
      }
    },
    {
      "DevicePath": "\\\\?\\hid#hid&col03#1&4784345&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "hanvonugee",
      "ProductName": "ugee device",
      "SerialNumber": "0000000000",
      "FriendlyName": "ugee device",
      "VendorID": 255,
      "ProductID": 47820,
      "InputReportLength": 10,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true,
      "DeviceAttributes": {
        "HID_REPORTS": "05:000D:0001"
      }
    },
    {
      "DevicePath": "\\\\?\\hid#hid&col05#1&4784345&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "hanvonugee",
      "ProductName": "ugee device",
[... 493 lines cut ...]
    {
      "Time": "2025-02-15T23:50:13.5554737+01:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2025-02-15T23:50:13.6755436+01:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

```

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

gonX, 2025-02-15:
How does this app expose itself? Currently we only support reading from USB devices directly.

Mrcubix, 2025-02-16:
Spacedesk uses its USB virtual device to send inputs to windows.
Whether or not it's sending tablets input or not, that can be figured from the HID Report Descriptor.
Would require a USB Dump to figure out which virtual endpoint data is received from.

quadrinex, 2025-02-16:
> How does this app expose itself? Currently we only support reading from USB devices directly.
> 
> /edit: If it doesn't use USB directly it will need a different backend, this is seen in [#1205](https://github.com/OpenTabletDriver/OpenTabletDriver/issues/1205) and [#1390](https://github.com/OpenTabletDriver/OpenTabletDriver/issues/1390)

it can use usb but i think wifi would work too