OpenTabletDriver/OpenTabletDriver #3814: Shift key continuously held down - XP-Pen Artist 16 2nd gen on Kubuntu

platinumtulip, 2025-02-20:
### Steps to Reproduce

1. plug in tablet
2. all pen strokes are registered as if Shift is being held down - layers are being dragged around etc.

### Expected Behavior

as far as I can tell, I do not currently have any modifier keys set to my shortcuts (auxiliary tablet shortcut keys, pen buttons, etc) and I am not holding down anything on my keyboard, either. pen strokes should be registered normally.

### Observed Behavior

so, just to be clear - I am not 100% sure yet if this is an issue with OpenTabletDriver, Kubuntu/KDE, or something else. but I tried my best to troubleshoot the issue on my own, and wanted someone else to take a look at it. I have used my XP-Pen Artist 16 with other OSes (Win11. LMDE) but only used their proprietary driver/GUI app for it. I wanted to switch to OpenTabletDriver to experiment with plugins, however. plus, the XP-Pen proprietary driver was giving me *other* issues on Kubuntu, like being unable to configure the active pen area properly... which OpenTabletDriver fixed for me, thankfully.

I've tried completely removing and reinstalling OpenTabletDriver, but the issue persists. I booted into LMDE (where I've only used XP-Pen's proprietary driver) and the tablet behaves as expected. I would really like to be able to port my drawing setup over to my Kubuntu install, since I significantly prefer KDE, but currently that's on pause while I try to figure out the root of this issue.

please let me know if this is not OpenTabletDriver's fault - I'm very sorry for the inconvenience if that is the case.

### Tablet Device

XP-Pen Artist 16 (2nd gen)

### Diagnostic Information

[xppen artist 13 2nd gen otd diagnostics.json](https://github.com/user-attachments/files/18879835/xppen.artist.13.2nd.gen.otd.diagnostics.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

jamesbt365, 2025-02-20:
This is more likely KDEs fault or anything like that if this issue is exclusive to KDE. 

KDE does its own weird tablet mapping and I wouldn't be surprised if a button is stuck somewhere in the input stack. OTD shouldn't directly be causing the keypress if nothing is bound to use it in the binding section.

A good usual way to check if its the desktop environment getting in the way is to see if it persists on absolute mode, you can switch to that at the bottom left of the main display in OpenTabletDriver.

If it works now, then your issue is external.

Also, diagnostics should not be 19MB, you are getting a lot of UI errors that should not be present in the logs from the daemon. Not sure why its like that but if you are going to export logs again, please close out OpenTabletDriver entirely first and restart the service (if you are using that).

platinumtulip, 2025-02-20:
good to know, thank you so much for the prompt response. also *wow*, I did not notice that the diagnostics were 19MB until you pointed that out. sorry about that!

unfortunately, switching to Absolute Mode still seems to trigger the issue. I'm still getting strange behavior in all programs as if Shift is being held down. I've attached an updated export log (now with a normal filesize) just in case that's helpful at all,  but I will start looking into how KDE might be messing things up here. 

[xppen16-2.json](https://github.com/user-attachments/files/18880094/xppen16-2.json)

gonX, 2026-10-04:
If your workflow includes unplugging/plugging your tablet regularly, #4989 which was just merged, should fix your issue. Please try a [Development Build](https://github.com/OpenTabletDriver/OpenTabletDriver/actions?query=branch%3A0.6.x+workflow%3A%22.NET%20CI%22) or wait for v0.6.8.