OpenTabletDriver/OpenTabletDriver #3827: Add support for Parblo N7B(F7)

nirajandata, 2025-03-03:
### Manufacturer

Parblo

### Product Name

Parblo Ninos N7B(F7)

### Device Width

111

### Device Height

177.5

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Diagnostic Information

[n7b.json](https://github.com/user-attachments/files/19056582/n7b.json)

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

jamesbt365, 2025-03-03:
You know, I would think after that discussion on discord you would realise that the name doesn't matter and that this tablet is already supported, you shouldn't use the configuration template directly, otherwise you're basically asking for your issue to be closed as a dupliacte and already completed.

I have restructured your issue this time, but you are responsible for you will be responsible for opening the right issue next time.

nirajandata, 2025-03-13:
It works fine on windows tho