OpenTabletDriver/OpenTabletDriver #3828: Support Request: Unit opentabletdriver.service does not exist

zamonary1, 2025-03-04:
### What do you need help with?

Yes, i've tried launching it as a user service!

I've installed opentabletdriver-git as an AUR using paru, and it has installed succesfully, but no user service is present in system.

```
╭─zamonary1@arch-pc ~  
╰─➤  systemctl --user enable --now opentabletdriver.service                                                                         1 ↵
Failed to enable unit: Unit opentabletdriver.service does not exist
```

I think that's the issue with paru, just installed it on a fresh system while distro switching.

### What have you tried?

Run it as a system service, and a user service

### Diagnostic Information

[image]

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

zamonary1, 2025-03-04:
Just tried it with yay, the issue is the same

jamesbt365, 2025-03-04:
Post the output of `ls /usr/lib/systemd/user` alongside the instructions you used to install OpenTabletDriver.

zamonary1, 2025-03-04:
Thank you for quick reply
[systemd/user](https://github.com/user-attachments/files/19078976/out.txt)
[instructions](https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Installation-Guide/174cc928f3725e5c7b4d2eb523b39be7ee6ca233#arch-linux-installation)

zamonary1, 2025-03-04:
Interstingly there IS such a service
Yes, i've tried reloading a daemon

jamesbt365, 2025-03-04:
Well, the wiki *was* migrated so you should be using https://opentabletdriver.net instead.

But, onto the actual issue, the service is actually present on your system so its not a problem with our packaging or anything and seems like a system issue, if the service file looks normal I'd try to refer to systemd documentation on how to reload stuff properly and check stuff like if its masked.

Not too familiar with stuff like this and does seem like an unusual system issue.

zamonary1, 2025-03-04:
Using the absolute path seemingly fixed the issue.
```
╭─zamonary1@arch-pc ~  
╰─➤  systemctl --user enable --now /usr/lib/systemd/user/opentabletdriver.service 
Created symlink '/home/zamonary1/.config/systemd/user/graphical-session.target.wants/opentabletdriver.service' → '/usr/lib/systemd/user/opentabletdriver.service'.
```

closed 2025-03-04 by jamesbt365